winder/Universal-G-Code-Sender #2524: Error while processing response <ok>: An unexpected command was completed by the controller.

my4040, 2024-05-21:
### Version

2.1.7

### Hardware / Firmware

Other (add details about it below)

### What happened

Hi, 
It's my first time using USG with Genmitsu 4040-pro. I was trying to run the machine using one of the sample files provided with the USB stick but the app is returning the entire time an error saying (see attached file for more details):

Error while processing response <ok>: An unexpected command was completed by the controller.

When I send the file to the CNC router it starts buzzing, it's trying to push the x achse back even if its already all the way at the end touching the y achse.

I would greatly appreciate your help!

[image]

breiler, 2024-05-22:
If you have an offline controller connected it must be disconnected while using UGS.

You will have to solve the errors before you can send a file.

my4040, 2024-05-22:
No, I don't have an offline controller. What else could cause the error?

my4040, 2024-05-22:
That's what the console is returning:

*** Connecting to jserialcomm://COM3:115200
*** Fetching device status
>>> ?
<Idle|MPos:0.000,0.000,0.000,0.000|Bf:35,254|FS:0,0|Pn:RHS|WCO:66.040,335.280,-76.204,0.000>
ok
>>> $I
[VER:ARM32 V3.0.20230816:]
[OPT:VZL,35,254]
ok
>>> $$
$0 = 10    (step pulse,usec)
$1 = 25    (step idle delay,msec)
$2 = 0    (stepport invert mask)
$3 = 4    (dirport invert mask)
$4 = 0    (stepenable invert,bool)
$5 = 1    (lims pin invert,bool)
$6 = 1    (probe pin invert,bool)
$10 = 115    (status report mask)
$11 = 0.010    (junction deviation)
$12 = 0.002    (arc tolerance,mm)
$13 = 0    (report inches,bool)
$20 = 0    (soft limits,bool)
$21 = 1    (hard limits,bool)
$22 = 1    (home cycle,bool)
$23 = 3    (homing dir invert mask)
$24 = 100.000    (homing feed,mm/min)
$25 = 500.000    (homing seek,mm/min)
$26 = 250    (homing debounce,msec)
$27 = 1.000    (homing pull-off,mm)
$30 = 10000    (maximum spindle speed,rpm)
$31 = 0    (minimum spindle speed,rpm)
$32 = 0    (laser mode enable,bool)
$100 = 800.000    (X axis pulse:step/mm)
$101 = 800.000    (Y axis pulse:step/mm)
$102 = 800.000    (Z axis pulse:step/mm)
$103 = 177.777    (A axis pulse:step/mm)
$110 = 2000.000    (X axis max rata:mm/min)
$111 = 2000.000    (Y axis max rata:mm/min)
$112 = 2000.000    (Z axis max rata:mm/min)
$113 = 1000.000    (A axis max rata:mm/min)
$120 = 20.000    (X axis acceleration:mm/s^2)
$121 = 20.000    (Y axis acceleration:mm/s^2)
$122 = 20.000    (Z axis acceleration:mm/s^2)
$123 = 20.000    (A axis acceleration:mm/s^2)
$130 = 400.000    (X aixs max travel:mm)
$131 = 400.000    (Y aixs max travel:mm)
$132 = 80.000    (Z aixs max travel:mm)
$133 = 9999.000    (A aixs max travel:mm)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
*** Connected to GRBL 3.0
>>> $H
ok
>>> G90 G0 X0 Y0
>>> G90 G0 Z0
ok
ok
>>> G10 P0 L20 X0 Y0 Z0
ok
>>> $H
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91X1F100
ok
>>> $J=G21G91Y-1F100
ok
>>> $J=G21G91Y-1F100
ok
>>> $J=G21G91Y-1F100
ok
>>> $J=G21G91Y-1F100
ok
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
An unexpected error was detected: (ALARM:3) Reset while in motion. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.
[MSG:Check Limits]
Error while processing response <ok>: An unexpected command was completed by the controller.

my4040, 2024-05-22:
BTW I was able to setup the machine following your James' instructions https://youtu.be/6V9tmybmdnY, however it started throwing the error messages again as soon as I configured the height of the Z using the Z probe.
After configuring the Z height I clicked return to Zero but it moved the spindle to the corner and it lowered it below the board. When I started the job, it started cutting from the edge and it went into a hold status.

I have attached few screenshots.
Thank you for your help!

[image]
[image]
[image]
[image]
[image]

breiler, 2024-05-22:
It is hard to follow what steps you are taking. 

From the console log it looks like you were:
1. Connecting
2. Homed the machine
3. Going to workspace zero
4. Reset workspace zero
5. Homed the machine again
6. Jogged the machine until it hit the limit switches

But it does not explain the "An unexpected command was completed by the controller". I have no idea what is going on.

my4040, 2024-05-23:
Hey Joacim, is there any online trainings that I can book or an option to get on a 1:1 session with you?

breiler, 2024-05-24:
@my4040 email me privately on joacim.breiler@gmail.com and let me know your time zone so that we can set something up.

breiler, 2024-05-27:
I am closing this as the machine seemed to be running fine during our session

closed 2024-05-27 by breiler

Telswoodworking, 2025-01-28:
Hi, I am have a very similar issue with "Error while processing response : An unexpected command was completed by the controller."

Right from when I connect, it says [MSG:Check Limits]
and then the error messsage.

Any help would be greatly appreciated.