winder/Universal-G-Code-Sender #2532: UGS dmg file not usable on MacOSX Sonoma 14.5 with MacBook Air / M2

mrRobot62, 2024-05-31:
### Version

2.1.7

### Hardware / Firmware

GRBL 1.1

### What happened

After downloading dmg-file for silicon M2 chip, I can't open the DMG-File. Nothing happens. Normally the security mechanism in OSX prevent installing 3rd party dmg. In Configuration/Security inside osx I do not get an information, that I try to install a 3rd party app.

It's not the first time, that I install dmg on this way.

Could somebody help me?

### How to reproduce

Reboot system, try installing again - same result - not possible

### Operating System

OSX Sonoma 14.5, MacBook Air M2

mrRobot62, 2024-06-09:
I found a solution how to install this corrupt dmg-file under OSX - and it works

1. copy dmg to you download folder
2. open a terminal and change to the download folder
3. you have to mount the dmg file
> `hdiutil attach macosx-aarch64-ugs-platform-app-2.1.7.dmg`
> now a new mount point is created
4. change to this new mount point
> `cd /Volumes/Universal\ Gcode\ Sender` 
5. copy the *.app file to your application folder
> `cp -r Universal\ Gcode\ Sender.app /Applications`
6. I renamed this new application to a name without `spaces` in the name ;-)
7. now you can start this new application directly from `Launchpad`

breiler, 2024-06-18:
I have found and fixed the broken link. But had no problems opening the DMG on an older iMac or a brand new Macbook Pro running Sonoma.

I am closing this as I am not able to reproduce it.

closed 2024-06-18 by breiler

mtwestra, 2024-09-08:
I had the same issue as @mrRobot62, with UGS version 2.1.8, on Sonoma 14.6.1. Clicking on the dmg does nothing. His solution worked for me.