winder/Universal-G-Code-Sender #2534: Autolevel map not showing after scan

marshr32, 2024-06-02:
### Version

2.1.7

### Hardware / Firmware

GRBL 1.1

### What happened

I've seen the old autolevel tutorial, and the heightmap preview that gets generated after scanning.
Even though I have the setting "show preview" enabled (both in the autoleveller and the visualizer settings), with the standard colours, I get nothing.
The autolevelling works, I can see the total height of the part change in the visualizer, but no red/green map is shown.

Is it that I can't see it because my part is 400mm across and my stock was 1.49mm max down from 0? ie the map is not very green or very red compared to the scale of the part?
Is there a way to amplify the colour changes?

I've also just noticed that my Z min and max settings have been changed from -5 and 5 which I used for the last scan, to -1.49 and 0, which are the results from that scan.
Is that meant to happen? 
It's gonna cause problems with the next scan if I don't change it every time.

### How to reproduce

I'm not sure if it's something to do with my PC, UGS, or just the scale of things.

Do a large scan with minimal out of level?

### Operating System

Edition	Windows 10 Home N Version	20H2 Installed on	‎2/‎03/‎2021 OS build	19042.1706 Experience	Windows Feature Experience Pack 120.2212.4170.0

breiler, 2024-06-02:
I am answering this from a sun bed in Sicily, so I am a bit limited in the help I can give.

The height map should render the colors from green which is the highest measured point to red as the lowest. Even if the difference is 0.01 mm you should be able to see it.

Is it possible that the gcode model is currently above the scanned surface?

The Z min/max is set when you press to take the min/max values from the gcode model.

Can you take a screen shot?

marshr32, 2024-06-02:
That's a very nice place to be! I loved Sicily when I went.

I'm about to start another run so I'll grab a screenshot, but I do all my models with the top of the stock as Z0, so it's all below the surface.

Another question which will help me in the short term so you can enjoy your holiday....
When the scan is done, where is Z0? ie does the autoleveller drop everything down below the highest point?
Or does it go above and below a previously set Z0?

I ask as I'm cutting plastic so using a Z probe in the spindle, Z0 that in the centre of the model, scanning with autolevel, then swapping to the cutting bit and rezeroing that to the work surface. So should I Z0 the cutter in the centre, where I Z0'd the probe? 
Or at the new highest point? And how would I find that?

Cheers.

marshr32, 2024-06-04:
[image]

marshr32, 2024-06-04:
First pic is the scan settings.
Second pic is without any file loaded (should I still see the scan grid though?)
Third is with a file opened and the scan applied

breiler, 2024-06-08:
Sorry for the late reply, I can see that you've disabled the grid. 
Did you perhaps disable the autolevel preview here:
[image]

marshr32, 2024-06-09:
Hi Will, no worries, it's not time sensitive.

Funny thing is I don't even have that option, see below.
[image]

I do have the option checked in the visualizer settings though
[image]

If you could give this bit a quick look that'd really help

> Another question which will help me in the short term so you can enjoy your holiday.... When the scan is done, where is Z0? ie does the autoleveller drop everything down below the highest point? Or [...]

Thanks!

breiler, 2024-06-09:
Ah, you need to show both the autoleveler window and the visualizer at the same time. As soon as you hide the autoleveler plugin it will automatically hide the leveling.

I have seen a lot of people using the probe module which will render the probe puck. And it was rendered even if they didn't use that function at the time (the window was minimized). So the best way I came up with was to not render the probe puck if the probe window wasn't active. To keep things consistent I did the same with the autoleveler.

Regarding your probing, having the Z0 in the center would make things difficult I believe. I would have the gcode zero on top of the material (working in Z negative space). That way, after a autoleveling you should be able to do a Z0 anywhere on the material and the rest of the probed points should be relative to that.

marshr32, 2024-06-09:
Awesome, that fixed it!
Thank you.
And now I get the option on right click

Sorry, I should have been more clear. By centre I just mean the centre of the circle where the axis cross is in the pics above.
Z0 is the top surface of the material.

So if I zero the Z probe at X0Y0 then do the scan, will that point by default be Z0 in the scan and everything else is above or below that?
So then I should also zero my tool length at X0Y0?

marshr32, 2024-06-09:
Since there is an option to turn the scan preview off, would it be possible change the hiding behavior to be able to have the tab in behind my visualizer tab as I had it?

Otherwise I'd need to open and close the auto-leveler module every time I want it so I can have my nice big visualizer back.

Not a game changer, but if it's possible........

breiler, 2024-06-09:
> So if I zero the Z probe at X0Y0 then do the scan, will that point by default be Z0 in the scan and everything else is above or below that? So then I should also zero my tool length at X0Y0?

Yes, that sounds correct

> Since there is an option to turn the scan preview off, would it be possible change the hiding behavior to be able to have the tab in behind my visualizer tab as I had it?

One possibility could be to have these features toggleable from this toolbar instead.
[image]

I'll have to think about it...

marshr32, 2024-06-09:
> 
> One possibility could be to have these features toggleable from this toolbar instead. 

That's a great idea.
And if the probe module could be toggled from there too that'd be amazing!

Thanks again for all you help and your work on the program.