winder/Universal-G-Code-Sender #2548: Falha na instalação

diogo-alves13, 2024-06-13:
### Version

2.1.7

### Hardware / Firmware

GRBL 1.1

### What happened

Tenho um notebook com windows 10 e quando tento instalar o UGR aparece uma mensagem com a seguinte informação "Não pode ser executado a partir de uma pasta que contém caracteres não ASCII no caminho"

### How to reproduce

Qual a incompatibilidade do instalador
[image]

### Operating System

Windows 10

breiler, 2024-06-13:
You get this error if any directory in the path contains "special" characters such as áéåäö. Move it to another folder.

diogo-alves13, 2024-06-13:
Em quais pastas devo procurar ?

diogo-alves13, 2024-06-13:
Fiz o teste com o mesmo arquivo em outro notebook e funcionou

breiler, 2024-06-15:
Any folder leading up to the folder that you placed UGS in. If you gave me the complete path I could point it out for you.

diogo-alves13, 2024-06-15:
Consegui resolver, a pasta área de trabalho estava no caminho porque uso o onedrive. Restaurei o caminho para padrão e funcionou.

Muito obrigado.

closed 2024-06-15 by diogo-alves13