winder/Universal-G-Code-Sender #2565: Stopped talking to X-Carve

PeanutsWoodshop, 2024-07-02:
### Version

2.1.7

### Hardware / Firmware

GRBL 1.1

### What happened

So I'm running UGS off a little NUC with a touchscreen and wireless keyboard. Spent half of yesterday getting everything setup and playing with designs in V-Carve. Everything was working fine (except that the scaling was completely wrong, but I was just gonna start from scratch.) However today when I turned everything on, UGS can't see the X-Carve anymore. NOTHING has changed. No new cables, none of that. But when I click on connect I get, "Error opening connection, could not connect to controller on port jserialcomm://COM3:115200" 

I did try swapping the usb port and the cable, to no avail. I've tried Machine Setup since I was gonna start from scratch any who but I can't get to it there either. But no matter what, I can't connect to the X-Carve like I did yesterday.

### How to reproduce

Unknown

### Operating System

Windows 11, version 10.0

breiler, 2024-07-02:
* Make sure you don't have an e-stop pressed on the controller.
* Do a cold reboot of your X-carve (make sure it is really powered off and the USB is unplugged)
* Check in the port drop down in UGS that port is still called COM3 and hasn't changed to something else

PeanutsWoodshop, 2024-07-02:
I don't want to talk about the fact that I'm surrounded by E-Stop buttons at work, but couldn't seem to spin the top far enough to get it to pop up. Should have had more caffeine this morning. Thank you!

breiler, 2024-07-02:
Glad it was that simple!

closed 2024-07-03 by breiler