winder/Universal-G-Code-Sender #2568: Manjaro / Arch linux : Error opening connection: Could not connect to controller on port jserialcomm://ttyUSB0

SvenMeyer, 2024-07-10:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

```
Product Version: Universal Gcode Platform 20240709
Java: 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9
Runtime: OpenJDK Runtime Environment 13.0.1+9
System: Linux version 6.9.5-1-MANJARO running on amd64; UTF-8; en_US (ugsplatform)
User directory: /home/sum/.ugsplatform
Cache directory: /home/sum/.cache/ugsplatform/var/cache
```

I installed candle2-git (V2.4.r34.2ad3200.59.d41d8cd) from the AUR just using the built in software manager. (as far as I remember I added myself `sum` as a user to the user group `tty`) ... and Candle2 has no problem to connect to my CNC 3018.

I also installed UGS from the AUR just using the built in software manager, but every time I click the [connect] button I get the error:
`Error opening connection: Could not connect to controller on port jserialcomm://ttyUSB0`

I also added myself to the group `uucp`, logged out and back again to be sure, checked the groups
```
$ groups sum
wheel lp tty uucp sys network power autologin sum
```
... however, no change, same error :-(

### Operating System

Linux version 6.9.5-1-MANJARO amd64

### Anything else

log file : [messages.log](https://github.com/user-attachments/files/16161749/messages.log)

SvenMeyer, 2024-07-10:
[UPDATE] I just found this old post 
[https://bbs.archlinux.org/viewtopic.php?id=178552](https://bbs.archlinux.org/viewtopic.php?id=178552)
which suggests to do (each time) also
`sudo chown $USER /dev/ttyUSB0`

Now I was able to connect and got some messages in the console window 
```
*** Connecting to jserialcomm://ttyUSB0:115200
>>> ?
ok
>>> $I
[VER:]
[OPT:V,15,511]
ok
>>> $$
$0 = 8    (Step pulse time, microseconds)
...
```

However when I wanted to execute a move command, I noticed that it UGS was not connected any more, and now when I clicked the `connect` button I could not reconnect and got the initial error message again ...

breiler, 2024-07-14:
Sorry for the late reply.

I think that you've been on the right track adding the right user roles. The group that you added to your user should have sufficed but it might needed a complete restart? 

The workaround that you did with `sudo chown $USER /dev/ttyUSB0` does basically the same thing but will not be persisted between restarts.

The response `[VER:]` from the controller tells me that this isn't a "original" GRBL and that the firmware has been modified by the vendor. We use this response to figure out what firmware version this is. Since the vendor has removed it we will assume that it is GRBL 1.1. Are you certain that it is GRBL 1.1 compatible and not GRBL 0.9?

Can you post your messages.log after you tried to connect and it got disconnected and the complete console log messages with verbose enabled (right click in the console and click "Enable verbose logging")?

breiler, 2024-07-29:
Closing as we didn't receive further updates from the user

closed 2024-07-29 by breiler