winder/Universal-G-Code-Sender #2569: Joystick options panel is broken on one machine; works on another

JamesHagerman, 2024-07-13:
### Version

2.1.7

### Hardware / Firmware

GrblHAL

### What happened

I have 2 linux laptops with UGS installed, both at 2.1.7. The joystick works great on one of them.

But on the second, the joystick options panel is empty apart from the text `Joystick` in the middle of the panel. After navigating to the Joystick panel, and closing the Options window, the app doesn't function quite right. For example, it is not possible to re-open the Options window.

### How to reproduce

Open the Options window and navigate to the Joystick options panel.

When the Joystick options window is first displayed, errors are reported in `messages.log`. Here is the relevant message:

```
java.lang.NullPointerException: Cannot invoke "com.willwinder.ugs.nbp.joystick.service.JoystickService.addListener(com.willwinder.ugs.nbp.joystick.service.JoystickServiceListener)" because "this.joystickService" is null
	at com.willwinder.ugs.nbp.joystick.JoystickOptionsPanel.<init>(JoystickOptionsPanel.java:63)
	at com.willwinder.ugs.nbp.joystick.JoystickOptionsPanelController.getPanel(JoystickOptionsPanelController.java:35)
	at com.willwinder.ugs.nbp.joystick.JoystickOptionsPanelController.getPanel(JoystickOptionsPanelController.java:24)
	at com.willwinder.ugs.nbp.lib.options.AbstractOptionPanelController.isValid(AbstractOptionPanelController.java:61)
	at org.netbeans.modules.options.TabbedController.isValid(TabbedController.java:126)
	at org.netbeans.modules.options.CategoryModel$Category.isValid(CategoryModel.java:394)
	at org.netbeans.modules.options.CategoryModel$Category.access$1300(CategoryModel.java:302)
	at org.netbeans.modules.options.CategoryModel.dataValid(CategoryModel.java:215)
	at org.netbeans.modules.options.OptionsPanel.dataValid(OptionsPanel.java:316)
	at org.netbeans.modules.options.OptionsDisplayerImpl$1$1.run(OptionsDisplayerImpl.java:234)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
[catch] at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:136)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
SEVERE [null]: Last record repeated more than 10 times, further logs of this record are ignored until the log record changes.
```

Because this works on one laptop but not the other, I assume there's a missing library or difference between the two systems... but I don't know enough about the code or the joystick dependencies to dig in, fix the issue, and submit a PR.

### Operating System

Broken on Ubuntu 18.04.06 LTS; works on PopOS! (ubuntu clone) 22.04 LTS

breiler, 2024-07-14:
Can you post the whole log file for the 18.04.06?

My guess is that this is a glibc incompatibility. Ubuntu Bionic (18.04) looks like it is shipped with glibc 2.27. The SDL library (which the gamepad depends on) is likely built with glibc 2.31.

If this is the case, this can't be fixed in UGS and would require you to upgrade your Ubuntu machine.

breiler, 2024-07-26:
Closing as we didn't get further response from user.

closed 2024-07-26 by breiler