winder/Universal-G-Code-Sender #2578: "!8 Err: Unknown command '?:3F'" while trying to connect to iDraw 1.0

thmsgbrt, 2024-07-27:
### Version

2.1.7

### Hardware / Firmware

GRBL 1.1

### What happened

Hello,

I'm trying to connect a iDraw 1.0 to UGS 2.1.8 on Mac (I've tried 2.1.7 and got the same result).
The only error I can find is the one on the screenshot.
I did not find any log in `~/Library/Application Support/ugsplatform/var/log/messages.log`

[image]

Thanks for the help!

### Operating System

Apple M1 Pro - Ventura 13.2.1

breiler, 2024-07-27:
From a quick search, the IDraw uses a different protocol (EBB) which is not supported by UGS.

thmsgbrt, 2024-07-27:
Hi Breiler,

Thanks for the quick answer.

I'm following a guide they published https://uunatek.com/blogs/tips-and-tricks/understanding-gcode-and-setting-up-the-idraw-machine-for-the-first-time-gcode-usage 

Perhaps my specific model is using the protocol you mentioned. Thank you!

breiler, 2024-07-27:
The responses in the screenshot is nothing that I recognize for any controller that UGS supports. You will need to contact your vendor for instructions on how to use your machine.

thmsgbrt, 2024-07-27:
Yes there are other ways to control it, I just hoped that UGS would work, as it would definitely be the best option ;) 

Thanks again for the help!

closed 2024-07-27 by thmsgbrt