winder/Universal-G-Code-Sender #2591: AutoLeveler Problems

Cani369, 2024-08-20:
### Version

Other

### Hardware / Firmware

GRBL 1.1

### What happened

As soon as i open ugs it throws me an error saying...
java.lang.NullPointerException: Cannot invoke "com.willwinder.ugs.platform.surfacescanner.MeshLevelManager.clear()" because "this.meshLevelManager" is null
	at com.willwinder.ugs.platform.surfacescanner.AutoLevelerTopComponent.componentClosed(AutoLevelerTopComponent.java:116)

And when i hit Autoleveler Pluggin it throws another error saying...
java.lang.NegativeArraySizeException: -2147483648
	at com.willwinder.ugs.platform.surfacescanner.SurfaceScanner.reset(SurfaceScanner.java:133)
	at com.willwinder.ugs.platform.surfacescanner.SurfaceScanner.update(SurfaceScanner.java:92)
	at com.willwinder.ugs.platform.surfacescanner.SurfaceScanner.<init>(SurfaceScanner.java:65)
	at com.willwinder.ugs.platform.surfacescanner.AutoLevelerTopComponent.componentOpened(AutoLevelerTopComponent.java:99)

The first time i downloaded it work fine, but all of the sudden it stop working.
Im using the 2.1.8 version.
If installed the 2.1.7  version also and got the same problem.

breiler, 2024-08-21:
I would like to see your .ugs/UniversalGcodeSender.json settings file to be able to reproduce this problem.

Depending on your operating system you can find it here:
* Windows: c:/users/[your username]/.ugs
* Mac: ~/Library/Preferences/ugs
* Linux: ~/.ugs

breiler, 2024-11-12:
Closing as we did not get further information from user.

closed 2024-11-12 by breiler