winder/Universal-G-Code-Sender #2595: Spindle not engaging

CaptainKokomo, 2024-08-31:
### Version

2.1.7

### Hardware / Firmware

GRBL 1.1

### What happened

I have had this issue repeatedly. However, previous fixes are not working this time. 
USG is controlling my genmitsu 6050 prover xl plus
I attached the laser (cfl55p) It's listed as not compatible with the 6050 but Sainsmart have told me it will work fine.
Once I disconnect the laser and reconnect the spindle, power is not being sent to the spindle. 
UGS still thinks its in laser mode for some reason. I have disabled laser mode. 
I have fresh installed to 2.1.8 today and same issue.
I deleted the cache folder, didn't fix it.
I have unplugged the laser cable from the control box.
The g-code is correct. There is a M3 command in there.
The control box just clicks when I manually send an M3 command, and clicks again with an M5 command. 
I have fixed this issue by running an earlier project and the spindle engaged. However, that's not working anymore. I have crawled through forums, groups, had 3 different AI trying to help... 
This happens no matter where the Gcode is generated. Carveco, freecad, inkscape.... 
There are no errors that come up. UGS thinks the machine is working properly. 
If you need more information please just let me know I am happy to provide what ever info you need.

### How to reproduce

I reproduce the problem by attaching the laser, then reattaching the spindle. 

### Operating System

Windows 11

breiler, 2024-08-31:
What is the output from the command `$$`?

CaptainKokomo, 2024-08-31:
[image]
I am not a sparky but I bought a cheap multi meter and power isn't being sent to the spindle even when the control box clicks with the M3 and M5 commands.

breiler, 2024-09-01:
Can you post your gcode file that you are trying to send?

What is the S-value together with the M3-command?

CaptainKokomo, 2024-09-02:
Sorry, not super familiar with Github. This is the start of the code of the last job I tried to run. I have tried to run code that has previously worked also. 

I have tried different values with the m3 command. 100, 1000, 12000... I'm open to suggestions.

T1
G17
G21
G90
G0Z32.6790
G0X0.0000Y0.0000
S12000M3
G0X-0.4690Y6.2330Z32.6790
G1F240.0
G1X-0.3527F800.0
X-0.0040
X0.1123Z-4.0742
X269.9352
X270.0515Z32.6790
X270.7490
Y6.4317
X270.0515
X269.9352Z-4.0742
X0.1123
X-0.0040Z32.6790
X-0.4690
Y6.6305
X-0.0040
X0.1123Z-4.0742
X269.9352
X270.0515Z32.6790
X270.7490
Y6.8292
X270.0515
X269.9352Z-4.0742
X0.1123
X-0.0040Z32.6790
X-0.4690
Y7.0279
X-0.0040
X0.1123Z-4.0742
X269.9352
X270.0515Z32.6790

breiler, 2024-09-02:
As far as I can tell this should work fine. You have laser mode disabled. Your max spindle speed setting is 12000. 

When issuing the gcode command `S12000M3` like it says in your file, it should turn on the spindle at max speed.

This sounds a hardware issue, you will have to contact Sainsmart support.

CaptainKokomo, 2024-09-02:
yes, I have recently got a replacement control boards and a replacement control box. The control box being used is very new. Are you saying it's a dodgy spindle?
I did have the spindle working before plugging the laser in.
When I bought the laser there was issues around the com ports and it needing separate com ports for laser grbl and for UGS... I did full on com port resets and fresh reinstalls of all software. The spindle eventually started working by running an older project. I did uninstall lasergrbl as well as the files were defaulting to laser files for what ever reason.

breiler, 2024-09-02:
All I know is that the software looks to be correctly configured and you are also sending the correct GCode. There is no way for me to further pin point the problem without access to the machine.