winder/Universal-G-Code-Sender #2599: Rasbperry Pi Visualizar works once, time when open a gcode; then no more...

davemaster, 2024-09-04:
### Version

2.1.8

### Hardware / Firmware

GRBL 0.9

### What happened

When I open UGS 2.1.8 (in Windows 10, and In Raspberry Pi 5), all works fine, I open a gcode file, it shown the lines into the visualizer, but, When I play to start doing the job, the lines, all of them in visualizer vanished, also but for a moment, when I connect the Arduino with firmware 0.9j (with PEN support). Searching, I found that the UGS 2.0.13( RPi 5), the platform version works perfect as shown in the picture bellow...

[image]

What should it be? Is the 2.0.13 version done for work with grbl 0.9j ??? 

Also works on Windows 10, the UGS Platform as shown bellow:

[image]

But with 2.1.8 (Raspberry Pi 5) and 2.1.6 (Windows 10) the failure shown in video bellow:

https://github.com/user-attachments/assets/9da6b23b-eedb-4d56-b31c-df9cccc69e48

Thanks in advance

### Operating System

Windows 10, Raspbian wayland (RPi 5)

breiler, 2024-09-04:
Can you give me a sample gcode file where this happens. And I also would like the log file (Go to the menu Tools -> Open log directory).

davemaster, 2024-09-04:
Ok, the log attached, is from the RPi 5, running UGS 2.1.8
Also, the gcode, but NO MATTER what gcode is, it happens; not with the
2.0.13 Platform, it does sweet...

Thanks in advance

PD:
My firmware, GRBL 0.9j - servo support (M3S15, AND M3S40)

breiler, 2024-09-04:
You need to attach files through github, email attachments doesn't not work.

davemaster, 2024-09-04:
> You need to attach files through github, email attachments does not work.

Ok,
here they are zipped
[rewinderuniversalgcodesenderrasbperrypivisualizer.zip](https://github.com/user-attachments/files/16875479/rewinderuniversalgcodesenderrasbperrypivisualizer.zip)

breiler, 2024-09-04:
The log contains why this problem occurs.

Can you right click in the console and choose "Enable verbose logging" and let it spew a couple of lines (then you can disable it again). Then copy them here, I just want see a couple of status report lines.

I would also like you to copy the output from the command `$$`.

davemaster, 2024-09-04:
> The log contains why this problem occurs.
>
> Can you right click in the console and choose "Enable verbose logging" and let it spew a couple of lines (then you can disable it again). Then copy them [...]

Ok, as soon as possible... Thanks

closed 2024-11-28 by breiler