winder/Universal-G-Code-Sender #2600: Grbl $ commands cannot be included in a Gcode file

GrahamJB, 2024-09-04:
### Version

2.1.8

### Hardware / Firmware

GRBL 1.1

### What happened

Grbl Laser mode is set by the $32=1 configuration command and turned off by $32=0 I use both Laser and spindle motor, obviously not at the same time. but I normally place a $32=1 at the top of a.nc file which uses a laser and $32=0 at the top of files that use the spindle. followed by a G4 P

When I use these files in UGS they are rejected as $32=0 or 1 is not a valid Gcode command although all Grbl boards will happily accept this, UGS is being overzealous in filtering out valid commands 

### How to reproduce

Put a GRBL setting command into a .nc file, it will be highlighted as an invalid line by UGS

### Operating System

Windows 10

### Anything else

Every time

breiler, 2024-09-04:
I would probably advice against using GRBL setting commands within a gcode file as these are not a part of GRBL:s motion planner and can have unexpected side effects. It will also lower the life span of the controller EEPROM.

I am guessing that you are getting an error message saying that the controller needs to be in an IDLE state to be able to execute the command. This is GRBL responding with this error and is not a UGS limitation.

If you still want this setting command to be executed within a program you can force the controller to catch up  by using a pause gcode command prior to the setting:
```
G4P0
$32=0
```

I will probably not change the syntax highlighting in the editor to allow `$`-commands. These are not gcode and should not be used as such.

GrahamJB, 2024-09-04:
I would agree generally about embedding them in a Gcode file, however I am talking at the top of a file, before any actual Gcode has been sent.

 

I see no difference (apart from the EEPROM Write time) between typing in a $32=1 in the console and then sending a Gcode file to putting the $32=1 at the top of the file.

 

This is definitely a UGS problem, the error is shown in UGS even if there is no router connected, Grbl is blissfully unaware of the start of a program, it just sees commands arrive over the USB cable. UGS won’t even attempt to send a command which Grbl will accept and execute.

 

If Grbl dislikes the data it receives it will generate errors itself.

breiler, 2024-09-04:
Having $32=0 in the start of a program worked just fine for me. It rendered ok, it displays an error in the editor (which I think it should) and it got sent as a part of the gcode program:

https://github.com/user-attachments/assets/b4bc8b59-2d31-40de-a7c5-ac91ffe0d62b

GrahamJB, 2024-09-04:
This is what I get, it looks like it has been sent OK but the error from Grbl is not shown, hence my assumption that it was UGS not sending it.

 

I have moved the $32=1 to the start in the code.

 

The error was only displayed when I sent a Jog command. Everything was OK until then. The error only was shown as returned only after the jog command. The router did not move from either the Gcode or Jog.

 

After a soft reset and the messages from a successful reset are displayed the console shows a message of **** Finished sending file in 00:00:27 *** 

 

I am no longer quite so certain that this is a UGS problem but I will run some more tests tomorrow, out of time for now!

 

https://github.com/user-attachments/assets/b4bc8b59-2d31-40de-a7c5-ac91ffe0d62b

GrahamJB, 2024-09-05:
Thank you for your responses,

 

I am coming to the conclusion that it is best not to mix Gcode with other Grbl commands.

However I will still use some $ commands in ‘macro’ files, for example when I change from Spindle to Laser and back I modify Laser mode and max and min spindle speeds and it is easier to place these into MachineSetLaserMode.nc and MachineSetSpindleMode.nc, can’t really use macros as the values are dependent on the specific router.

 

UGS is this morning being very flaky connecting to the router, and still will not return an error until a jog movement is made (most of the time, sometimes it works as it should), before you mention connections and cables I am well aware of these possible problems, Candle and Gsender do not have these problems using the same router, cable and Com Port.

 

If I can track down any pattern for the inconsistency I will let you know.

 

Best Regards

Graham Bland

 

https://github.com/user-attachments/assets/b4bc8b59-2d31-40de-a7c5-ac91ffe0d62b

breiler, 2024-09-05:
Ok, I am closing this then.

Feel free to create a new issue with the connection problem if you find something that might be reproducable.

closed 2024-09-06 by breiler