winder/Universal-G-Code-Sender #2604: UGS wont connect under FluidNC drop down

itsnotlevel1, 2024-09-08:
### Version

2.1.8

### Hardware / Firmware

FluidNC

### What happened

UGS Wont connect to machine using the FluidNC drop down, it keeps telling me...
*** Fetching device settings
*** Could not establish connection with the controller
*** Connection closed
*** Could not establish connection with the controller

But when I choose grbl esp32 from the drop down, it connects and runs fine. Runs a job if I load it. 
I tried everything thinking it was a hardware issue, its not a hardware issue that I can see. 

### How to reproduce

If I choose FluidNC from drop down, it will not connect and gives the same response. But grbl esp32 connects every time and works with the machine. 

### Operating System

Windows 11 Pro

### Anything else

Every time.

breiler, 2024-09-09:
What is the output of `$I`?

itsnotlevel1, 2024-09-09:
[VER:3.8 FluidNC v3.8.2:]
[OPT:MPHS]
[MSG:Machine: xPro V5 ]
[MSG:Mode=STA:SSID=Garage3_EXT:Status=Connected:IP=10.0.0.122:MAC=8C-4B-14-BB-D6-6C]
ok

breiler, 2024-09-10:
Worked just fine for me:
[image]

What is the controller state after you connected using the GRBL_ESP32 profile?

Can you disconnect, enable verbose logging (right click in console and select "Enable verbose logging") and connect again with the FluidNC profile and paste the messages here.

Can you attach your config.yaml.

itsnotlevel1, 2024-09-11:
After Connecting to grblesp32 profile, the machine and controller are ready to be used. Ive run a program with it. 

Heres the output of Fluidnc connection profile:
*** Connecting to jserialcomm://COM3:115200
Other: ets Jun  8 2016 00:22:57
Other: 
Other: rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
Other: configsip: 0, SPIWP:0xee
Other: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
Other: mode:DIO, clock div:1
Other: load:0x3fff0030,len:1184
Other: load:0x40078000,len:13260
Other: load:0x40080400,len:3028
Other: entry 0x400805e4
INFO: uart_channel0 created
RST
INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC
INFO: Compiled with ESP32 SDK:v4.4.7-dirty
INFO: Local filesystem type is littlefs
INFO: Configuration file:config.yaml
INFO: Machine xPro V5 
INFO: Board XPro V5
INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19
INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000
INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms
INFO: Axis count 3
INFO: Axis X (-1264.000,0.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050
INFO:  X Neg Limit gpio.35
INFO: Axis Y (-1289.000,0.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26:low CS:NO_PIN Disable:NO_PIN Index:2 R:0.050
INFO:  Y Neg Limit gpio.34
INFO:   Motor1
INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32:low CS:NO_PIN Disable:NO_PIN Index:3 R:0.050
INFO: Axis Z (-127.000,0.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050
INFO:  Z Neg Limit gpio.39
INFO: X Axis driver test passed
INFO: Y Axis driver test passed
INFO: Y2 Axis driver test passed
INFO: Z Axis driver test passed
INFO: safety_door_pin gpio.16
INFO: macro0_pin gpio.13:low
INFO: macro1_pin gpio.0:low
INFO: Kinematic system: Cartesian
INFO: Connecting to STA SSID:Garage3_EXT
INFO: Connecting.
*** Fetching device status
> ?
[MSG:INFO: Connecting..]
INFO: Connected - IP is 10.0.0.122
INFO: WiFi on
INFO: Start mDNS with hostname:http://fluidnc.local/
INFO: HTTP started on port 80
*** Fetching device status (2 of 10)...
> ?
[MSG:INFO: Telnet started on port 23]
INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: Mist coolant gpio.21
INFO: Probe gpio.22:low
Other: 
Grbl 3.8 [FluidNC v3.8.2 (noradio) '$' for help]
*** Fetching device status
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-1238.600,-1250.900,-76.200>
Other: ok
[... 566 lines cut ...]
$100 = 160.000    (X-axis travel resolution, step/mm)
$112 = 2000.000    (Z-axis maximum rate, mm/min)
$111 = 4000.000    (Y-axis maximum rate, mm/min)
$110 = 4000.000    (X-axis maximum rate, mm/min)
$122 = 250.000    (Z-axis acceleration, mm/sec^2)
$121 = 250.000    (Y-axis acceleration, mm/sec^2)
$120 = 250.000    (X-axis acceleration, mm/sec^2)
$132 = 127.000    (Z-axis maximum travel, millimeters)
$131 = 1289.000    (Y-axis maximum travel, millimeters)
$130 = 1264.000    (X-axis maximum travel, millimeters)
$10 = 1    (Status report options, mask)
$30 = 24000    (Maximum spindle speed, RPM)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
*** Connected to GRBL 3.8
[MSG:INFO: Huanyang PD014 Accel:5.000]
[MSG:INFO: Huanyang PD015 Decel:5.000]
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>

Yaml file: 
[config.yaml.txt](https://github.com/user-attachments/files/16956561/config.yaml.txt)

breiler, 2024-09-11:
I loaded up your config to a controller and tried it. I don't have TMC drivers or the Huangyang spindle so I am not able to simulate your scenario fully.

But I think that this is a clue:
```
INFO: Connecting.
*** Fetching device status

?
[MSG:INFO: Connecting..]
INFO: Connected - IP is 10.0.0.122
INFO: WiFi on
INFO: Start mDNS with hostname:http://fluidnc.local/
INFO: HTTP started on port 80
*** Fetching device status (2 of 10)...
?
[MSG:INFO: Telnet started on port 23]
INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: Huanyang Spindle Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: Mist coolant gpio.21
INFO: Probe gpio.22:low
Other:
Grbl 3.8 [FluidNC v3.8.2 (noradio) '$' for help]
*** Fetching device status
?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-1238.600,-1250.900,-76.200>
Other: ok
```

It looks like the controller has not completed the boot before UGS starts quering the controller. Can you connect and power up your controller and wait a minute before connecting to the controller through UGS. This should not be a problem, but looking through the lines after there are some wierd things happening.

The reason it fails to connect later on looks to be caused by it either getting duplicated messages from the controller . Here is an example, it receives the line with `180: Parameter Assignment Failed` and then an ok. But these two lines is later repeated.

```
175: Expression Syntax Error
176: Flow Control Syntax Error
177: Flow Control Not Executing Macro
178: Flow Control Out of Memory
179: Flow Control Stack Overflow
180: Parameter Assignment Failed
ok
Other: 180: Parameter Assignment Failed
Other: ok
```

itsnotlevel1, 2024-09-12:
So, I did what you asked and still nothing. Then I unplugged my vfd and it connected but nothing has changed unless for some reason theres a variable in the huanyang vfd I need to change? I havent changed anything. I unplugged and reconnected several times....it works now. But if I plug the spindle in, it wont connect. 
And as it should, when spindle unplugged it gives me this: INFO: VFD RS485 Unresponsive but thats typical.

breiler, 2024-09-12:
I noticed these two lines though when you connected to it using the GRBL profile and remember that there was another issue where UGS would try to initiate the connection twice on certain device setups (think it was when connecting to it via WiFi):
```
Grbl 3.8 [FluidNC v3.8.2 (noradio) '$' for help]
Grbl 3.8 [FluidNC v3.8.2 (noradio) '$' for help]
```

I fixed that particular issue in version 2.1.9, you could give that version a try:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.1.9

Other than that this will be really hard for me to debug and fix as I don't have a controller equipped with RS485 and VFD supporting it.

breiler, 2024-10-25:
Sorry for the late reply, another user had the same problem and I think that I've fixed it.

Please try the latest nightly build from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

closed 2025-02-15 by breiler