winder/Universal-G-Code-Sender #2611: Error on OrangePi

KuleshoffM, 2024-09-16:
### Version

2.1.8

### Hardware / Firmware

GRBL 1.1

### What happened

orangepi@orangepizero2w:/cnc/UniversalGcodeSender$ java -jar -Xmx256m UniversalGcodeSender.jar
Exception in thread "main" java.awt.HeadlessException:
No X11 DISPLAY variable was set,
but this program performed an operation which requires it.
        at java.desktop/java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:164)
        at java.desktop/java.awt.Window.<init>(Window.java:553)
        at java.desktop/java.awt.Frame.<init>(Frame.java:428)
        at java.desktop/java.awt.Frame.<init>(Frame.java:393)
        at java.desktop/javax.swing.JFrame.<init>(JFrame.java:180)
        at com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java:110)
        at com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:218)

### Operating System

Ubuntu

### Anything else

orangepi@orangepizero2w:/cnc/UniversalGcodeSender$ java -version
openjdk version "17.0.12" 2024-07-16
OpenJDK Runtime Environment (build 17.0.12+7-Ubuntu-1ubuntu222.04)
OpenJDK 64-Bit Server VM (build 17.0.12+7-Ubuntu-1ubuntu222.04, mixed mode, sharing)

breiler, 2024-09-16:
What is your desktop environment?

KuleshoffM, 2024-09-16:
I'm trying to run UGS on OrangePi. I installed Ubuntu there in terminal mode. There is no desktop there.

breiler, 2024-09-16:
How are you planning to run gcode files?

There is a cli-alternative, look in the ugs-cli.sh

KuleshoffM, 2024-09-16:
from web...
orangepi:8080

where look ugs-di.sh ?

breiler, 2024-09-16:
Sorry, it is named "start-cli.sh" and should be available in the UniversalGcodeSender.zip

KuleshoffM, 2024-09-16:
I was running start-cli.sh . There are no errors, but nothing happens. The web interface is also not available

breiler, 2024-09-16:
Run `start-cli.sh -h` to print help information.

You will need to run `start-cli.sh -d` to start in a daemon mode which should start the web server.

KuleshoffM, 2024-09-16:
Its work. Sencs

closed 2024-09-16 by KuleshoffM