winder/Universal-G-Code-Sender #2624: UGS will not connect under NCFluid. Problem seems to be connected to Haunyang VFD

Toolman19, 2024-10-24:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

Since installing NCFluid UGS will not connect anymore when NCFluid is selected but will connect okay when GRBL is selected. 
The problem seems to be connected to the Haunyang VFD with the error message but I'm not sure what I need to change.
Any advise would be appreciated.
I have included the UGS error message and my config file below.

*** Connecting to jserialcomm://COM3:115200
*** Fetching device status
*** Fetching device status
> 
ok
*** Fetching device firmware version
*** Fetching device status codes
*** Fetching device state
*** Fetching device settings
*** There was an error while reading the configuration, see detailed error message below:
while scanning a simple key
 in 'string', line 121, column 1:
    [MSG:INFO: Huanyang PD014 Accel: ... 
    ^
could not find expected ':'
 in 'string', line 122, column 9:
            tpfd: 4
            ^
"*** Could not establish connection with the controller
*** Connection closed
*** Could not establish connection with the controller

board: XPro V5
name: xPro V5 Ondrej 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 199.999
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 775.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 200.976
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 750.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.5000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 200.577
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 125.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 10000000

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: gpio.16:high
  macro0_pin: gpio.13:low
  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0

uart1:
  txd_pin: gpio.4
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  baud: 9600
  mode: 8N1
 
Huanyang:
  Uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
  

### Operating System

Windows 10 64bit

breiler, 2024-10-24:
Can you activate the verbose logging in the console and post the complete log from when you are connecting here.

Toolman19, 2024-10-24:
*** Connecting to jserialcomm://COM3:115200
Other: ets Jun  8 2016 00:22:57
Other: 
Other: rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
Other: configsip: 0, SPIWP:0xee
Other: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
Other: mode:DIO, clock div:1
Other: load:0x3fff0030,len:1184
Other: load:0x40078000,len:13260
Other: load:0x40080400,len:3028
Other: entry 0x400805e4
INFO: uart_channel0 created
RST
INFO: FluidNC 3.8.4-pre6 https://github.com/bdring/FluidNC
INFO: Compiled with ESP32 SDK:v4.4.7-dirty
INFO: Local filesystem type is littlefs
INFO: Configuration file:config.yaml
INFO: Machine xPro V5 Ondrej 
INFO: Board XPro V5
INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19
INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000
INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms
INFO: Axis count 3
INFO: Axis X (-625.000,150.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050
INFO:  X Neg Limit gpio.35:low
INFO: Axis Y (-600.000,150.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050
INFO:  Y Neg Limit gpio.34:low
INFO:   Motor1
INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050
INFO: Axis Z (25.000,150.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050
INFO:  Z Neg Limit gpio.39:low
INFO: X Axis driver test passed
INFO: Y Axis driver test passed
INFO: Y2 Axis driver test passed
INFO: Z Axis driver test passed
INFO: safety_door_pin gpio.16
INFO: macro0_pin gpio.13:low
INFO: macro1_pin gpio.0:low
INFO: Kinematic system: Cartesian
INFO: STA SSID is not set
INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1
INFO: AP started
INFO: WiFi on
INFO: Captive Portal Started
INFO: HTTP started on port 80
*** Fetching device status
> ?
[MSG:INFO: Telnet started on port 23]
INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: Mist coolant gpio.21
INFO: Probe gpio.22:low
Other: 
Grbl 3.8 [FluidNC 3.8.4-pre6 (wifi) '$' for help]
*** Fetching device status
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-214.512,-308.900,85.077>
Other: ok
> 
ok
*** Fetching device firmware version
> $I
[VER:3.8 FluidNC 3.8.4-pre6:]
[OPT:MPHS]
[... 447 lines cut ...]
  off_on_alarm: false
  atc: 
  m6_macro: 

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
ok
Other: ok
*** There was an error while reading the configuration, see detailed error message below:
while scanning a simple key
 in 'string', line 112, column 1:
    [MSG:INFO: Huanyang PD014 Accel: ... 
    ^
could not find expected ':'
 in 'string', line 113, column 9:
            toff_stealthchop: 5
            ^
"*** Could not establish connection with the controller
*** Connection closed
*** Could not establish connection with the controller

breiler, 2024-10-25:
Thanks

Try the latest nightly build from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

Toolman19, 2024-10-25:
I have downloaded the night build and tried connecting it to my controller which seems better and now seems to complete but is stuck in a loop.

*** Connecting to jserialcomm://COM3:115200
Other: ets Jun  8 2016 00:22:57
Other: 
Other: rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
Other: configsip: 0, SPIWP:0xee
Other: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
Other: mode:DIO, clock div:1
Other: load:0x3fff0030,len:1184
Other: load:0x40078000,len:13260
Other: load:0x40080400,len:3028
Other: entry 0x400805e4
INFO: uart_channel0 created
RST
INFO: FluidNC 3.8.4-pre6 https://github.com/bdring/FluidNC
INFO: Compiled with ESP32 SDK:v4.4.7-dirty
INFO: Local filesystem type is littlefs
INFO: Configuration file:config.yaml
INFO: Machine xPro V5 Ondrej 
INFO: Board XPro V5
INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19
INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000
INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms
INFO: Axis count 3
INFO: Axis X (-625.000,150.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050
INFO:  X Neg Limit gpio.35:low
INFO: Axis Y (-600.000,150.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050
INFO:  Y Neg Limit gpio.34:low
INFO:   Motor1
INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050
INFO: Axis Z (25.000,150.000)
INFO:   Motor0
INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050
INFO:  Z Neg Limit gpio.39:low
INFO: X Axis driver test passed
INFO: Y Axis driver test passed
INFO: Y2 Axis driver test passed
INFO: Z Axis driver test passed
INFO: safety_door_pin gpio.16
INFO: macro0_pin gpio.13:low
INFO: macro1_pin gpio.0:low
INFO: Kinematic system: Cartesian
INFO: STA SSID is not set
INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1
INFO: AP started
INFO: WiFi on
*** Fetching device status
> ?
null
INFO: HTTP started on port 80
INFO: Telnet started on port 23
INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600
INFO: Mist coolant gpio.21
INFO: Probe gpio.22:low
Other: 
Grbl 3.8 [FluidNC 3.8.4-pre6 (wifi) '$' for help]
*** Fetching device status
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-214.512,-308.900,85.077>
Other: ok
> 
ok
*** Fetching device firmware version
> $I
[... 1398 lines cut ...]
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-214.512,-308.900,85.077>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-214.512,-308.900,85.077>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
*** Connection closed

breiler, 2024-10-25:
That is just UGS polling the controller for status. Right click in the console and disable verbose logging.

Toolman19, 2024-10-25:
Just done that and its now connecting perfectly. Thank you very much for sorting this so quickly.

closed 2024-10-25 by Toolman19