winder/Universal-G-Code-Sender #2625: arduino chino, fallas con UGS sobre mala conexión con el arduino

PrototiposYPlacas, 2024-10-24:
### Version

2.1.8

### Hardware / Firmware

GRBL 1.1

### What happened

HOLA buenas tardes, necesito ayuda con la comunicación del Arduino al UGS de mi cnc. el problema esta en que cuando quiero probar los ejes para ver su movimiento, hay momentos que si lo hace y otras veces me entrega una alarma o se queda trabado.
"Cabe mencionar que tengo un Arduino chino". Mi profesor me dijo que talvez necesite un driver para que la comunicación con UGS sea estable ya le instale el driver CH340 para que mi PC detecte el Arduino, estoy atento a sugerencias por que de verdad llevo 2 semanas echando humo 

### How to reproduce

para reproducir el problema, necesitas un arduino chino+GRBL1.1v+CNC SHIELD + A4988 y UGS 2.1.8

### Operating System

Windows 11

PrototiposYPlacas, 2024-10-24:
RESUMIENDO necesito saber si me hace falta un driver para colocarlo en el UGS

breiler, 2024-10-25:
If you already have the driver installed and are able to connect to the Arduino controller from UGS, you are all set this far.

I can't help you further in investigating without knowing which error you are receiving, how you've set up your machine or how the controller is configured (the settings from `$$`).

breiler, 2024-11-12:
Closing as we did not get further information from user.

closed 2024-11-12 by breiler

PrototiposYPlacas, 2024-12-24:
hola como estas disculpa que no conteste, el problema que tenia lo resolvi
resulta y acontece que mi cnc Shield estaba mala la tuve que hacer desde
cero