winder/Universal-G-Code-Sender #2630: AutoLeveler - Scan surface fails when probing 3rd point

vinixtito, 2024-10-29:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

I'm using a lodaded file and setting up correctly the grid of points to scan, using the available AutoLeveler plugin commands. 
While doing the Scan surface, it stops the process on the third point, every time. A popup appears saying "Probe failed".
First I was doing the scan as it should be, but to isolate the issue I've tried touching manually every time the probing down fase starts, and it always stop on that 3rd point. The log shows no errors as you can see below

>>> G21G90G0Z20.2F1000
>>> G90 G21
>>> G21G90G0X0Y0F1000
>>> G90 G21
>>> G21G90G0Z15.2F1000
>>> G90 G21
>>> G21G90G0X0Y0F1000
>>> G90 G21
>>> G21 G91 G49
>>> G38.2 Z-15.2 F10
>>> G90 G21
ok
ok
ok
ok
ok
ok
ok
ok
ok
[PRB:-682.007,-725.499,-40.786:1]
ok
ok
>>> G21G90G0Z15.2F1000
>>> G90 G21
>>> G21G90G0X0Y70F1000
>>> G90 G21
>>> G21 G91 G49
>>> G38.2 Z-15.2 F10
>>> G90 G21
ok
ok
ok
ok
ok
[PRB:-682.007,-655.491,-40.838:1]
ok
ok
>>> G21G90G0Z15.2F1000
>>> G90 G21
>>> G21G90G0X0Y140F1000
>>> G90 G21
>>> G21 G91 G49
>>> G38.2 Z-15.2 F10
>>> G90 G21
ok
ok
ok
ok
ok
[PRB:-682.007,-585.483,-40.854:1]
ok
ok

### How to reproduce

Fisrt I set the machine to the home position, then do a Return to zero, and then I do the Scan surface. In each point when the probe starts the slower down movement, I deliberately touch the tool with the probe plate so it goes to the next point and repeat the process.

### Operating System

Windows 11 Home Edition Version 23H2 Installed on ‎11/‎10/‎2022 OS Build 22631.4317 Serial Number PF3LT6WC Experience Windows Feature Experience Pack 1000.22700.1041.0

breiler, 2024-10-29:
Can you post your log file. Open the Tools menu and choose "Open log directory", then upload the ´messages.log´ here.

vinixtito, 2024-10-29:
I made a video of the process described.
https://github.com/user-attachments/assets/5bbaf585-e0aa-4835-a95d-3b1f202941df

vinixtito, 2024-10-29:
[messages.log](https://github.com/user-attachments/files/17562203/messages.log)

breiler, 2024-10-29:
Thanks, this is an old problem that I have tried adjusting so many times.

There is a check that will make sure that the requested XY-cordinate to probe will match the reported XY-coordinate after the probe.

In your case UGS sent to probe the coordinate [X0, Y140] but the returned coordinate was [X0.003, 140.01]. As far as I know, this is caused by a rounding error on the controller. With greater steps/mm setting this problem will go away.

This check is clearly not working, so I will remove it.

breiler, 2024-10-29:
The change is merged and available in the latest nightly build:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

vinixtito, 2024-10-29:
Thanks for the rapid  answers. Since I’m in a different timezone I’ll only be able to test the new version tomorrow. Best regards!

vinixtito, 2024-10-30:
Couldn't test the Scan surface beacause file opening gives an error "No file loaded" in the status bar.

breiler, 2024-10-30:
Give me a a couple of minutes. I pushed a new feature today and it had a couple of problems with the packaging.

breiler, 2024-10-30:
I have reverted my latest changes, please try it now

vinixtito, 2024-10-30:
Sorry, but I can’t do it so  soon. Propably, only tomorow

breiler, 2024-10-30:
No worries, sorry for the inconvenience!

vinixtito, 2024-10-31:
Done. It's working. Thank you!

closed 2024-10-31 by vinixtito