winder/Universal-G-Code-Sender #2635: Error streaming to GRBL v1.1

hahnpv, 2024-11-08:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

"An error was detected while sending 'G2X-3.185Y-78.449I0J0.589': (error:1) G-code words consist of a letter and a value. Letter was not found. Streaming has been paused."

If I copy the gcode file to a SD card and use the offline controller on the CNC, code runs just fine.

Sending the above string in its original formatting over USB will replicate the issue:

G2 X-3.185 Y-78.449 I0 J0.589

thank you for your help,
philip

### How to reproduce

The gcode above will reproduce the problem

breiler, 2024-11-08:
Can you post the whole Gcode file. The G2 commands needs a starting point in order to function.

hahnpv, 2024-11-09:
Ah, sorry I overwrote the code trying to debug.
I generated a similar toolpath (facing operation) and hit the same error on a different line.

The entire gcode is attached. It trips on line 34.

[1001.nc.txt](https://github.com/user-attachments/files/17685779/1001.nc.txt)

When running the gcode from a SD card in the offline controller it executes successfully.

(PROverXL 6060, grbl1.1)

thank you,
philip

> **** Pausing file transfer. ****
>
> An error was detected while sending 'G1X69.835': (error:36) Unused value words found in block. Streaming has been paused.
> **** The communicator has been paused ****
> 
> **** Pausing file transfer. ****
> 
> An error was detected while sending 'G18G3X69.2Z-0.365I0K0.635': (error:2) Missing the expected G-code word value or numeric value format is not valid. Streaming has been paused.
> **** The communicator has been paused ****
> 
> **** Pausing file transfer. ****
> 
> An error was detected while sending 'G1Z5': (error:25) Repeated g-code word found in block. Streaming has been paused.

breiler, 2024-11-09:
I tried your file without any problems. Since you first pointed out an arc command I suspected that this was caused by a precision error which is common when the machine uses low steps/mm settings. This would force the controller to expand the arc to line segments with bad precisions making the controller throw an error.

But given the list of random errors you provided I suspect bad communication between the computer and controller where data is being either scrambled or dropped.

Make sure that you are using a properly shielded USB cable without any extensions or USB hubs and that it is placed far from any high power cables that may cause EMI (such as spindle or vaccum power cables).

To debug, you can try if this also happens if any other high power peripherals are turned off. For instance, running without the spindle.

breiler, 2024-12-05:
Closing as we did not get further responses from user

closed 2024-12-05 by breiler