winder/Universal-G-Code-Sender #2636: plese edit help about text

annopnod, 2024-11-10:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

this text is netbeans ide not ugs
[image]

breiler, 2024-11-10:
The feature in Netbeans for overriding this text got broken a couple of versions ago. There is currently no way for us to change this text. One option is to implement our completely own about dialog, but I think that it had a very low priority.

breiler, 2025-02-15:
We are not able to fix this until it is fixed upstream in netbeans platform. Closing as won't do.

closed 2025-02-15 by breiler