winder/Universal-G-Code-Sender #2637: support lasergrbl

annopnod, 2024-11-10:
### Description

support lasergrbl exsimple
open/close laser 1% for laser positioning.
alarm before run gcode.
lock z prevent loss focus.

breiler, 2024-11-10:
It is not very clear what you want from this request.

I can understand the feature about turning on the laser at minimal power that can be used for exact positioning of the machine.

I don't understand the alarm, lock z and loss of focus.

annopnod, 2024-11-11:
alarm before run gcode file.
Laser is a highly concentrated light.
May cause blindness from reflecting objects.
Reminder to wear protective glasses.

breiler, 2024-11-11:
> lock z is not interpreter z axis in gcode. In adjustable focus. z axis value have effect focus for cutting project .

So you want to be able to block any Z coordinate being sent to the controller? What are you using as CAM, why are you not disabling the Z axis there?

annopnod, 2024-11-12:
oh sorry.
i didn't expect it.
i use FlatCAM make pcb.