winder/Universal-G-Code-Sender #2642: Visualizer is not working

Kibris1977, 2024-11-20:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

I create a new design and put in the cutting requirements but the visualizer dose respond and i cant send to the machine. 

### How to reproduce

using the new design function create and design, input the cutting requirements.  The visualizer should show the model of the design but in the visualiser window there is nothing apart from the view cube.

### Operating System

windows 11

### Anything else

I have tried reinstalling but its still the same.

breiler, 2024-11-20:
Can you attach the ugsd-file for your project?

Franckapik, 2024-11-30:
I have also empty visualiser sometimes when i change my windows apparence...

breiler, 2024-12-03:
Closing as we did not get any response from user

closed 2024-12-03 by breiler