winder/Universal-G-Code-Sender #2646: Visualizer weirdness and freeze during startup

winder, 2024-11-30:
### Version

2.1.10

### Hardware / Firmware

GRBL 1.1

### What happened

On a fresh installation of 2.1.10 UGS started normally and the visualizer works as expected. I dragged it to the main view area and it stopped working, displaying only the netbeans window and toolbar plus the gray background. At this point 
closing the visualizer and re-opening it from the top menu fixes the display issue.

After the first start, if the visualizer is part of the layout the UI fails to render:

With the visualizer enabled, restarting UGS causes it to freeze completely and fail to render any UI components.
[image]

### How to reproduce

Start with the `ugsplatform` shell script.

* first start: works normally
* second start: completely frozen with no UI rendered if visualizer is part of the layout.
* moving visualizer view location: openGL window appears to go away completely until the view is closed and re-opened.

### Operating System

Linux, PopOS, AMD Ryzen 7 7840U w/ Radeon  780M Graphics, AwesomeWM

### Anything else

messages.log looks normal except for this at the end:
```
SEVERE [global]
java.lang.RuntimeException: Waited 5000ms for: <2e6951a9, 20aeaa14>[count 1, qsz 0, owner <AWT-EventQueue-0-Display-.x11_:1-1-EDT-1>] - <AWT-EventQueue-0-FPSAWTAnimator#00-Timer0>
	at jogamp.common.util.locks.RecursiveLockImpl01Unfairish.lock(RecursiveLockImpl01Unfairish.java:198)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:793)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
Caused: com.jogamp.opengl.util.AnimatorBase$UncaughtAnimatorException
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:92)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:471)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:245)
	at java.base/java.util.TimerThread.mainLoop(Unknown Source)
[catch] at java.base/java.util.TimerThread.run(Unknown Source)
```

additional troubleshooting
---

Downgrading to 2.1.9 allows everything to start normally. I'm able to close the visualizer window and switch back to 2.1.10.

Deleting the .ugsplatform directory also allows me to start 2.1.10 again.

breiler, 2024-11-30:
Thanks for testing! I have noticed similar problems while developing where I shut down the Java process from the IDE, but only from my workstation running an NVidia graphics card.

I have never experienced it when running it "normally", so I figured that I could live with it as a developer. I have not experienced this problem on either MacOSX or Windows.

I will have to do some more testing before hitting the publish button.

breiler, 2024-12-13:
Damn it! =/

I'll see if I can reproduce it somehow. I have tested it on all the machines that I have available and it has worked without any problems on all of them (Linux, Windows, MacOSX AARCH64, MacOSX Intel).

I really think this in the end will improve the user experience, so I think that it is worth trying to get this to run.

That other machine that you have sounds like it is quite beefy. Since the original problem was with deadlocks due to the timing of different threads, that machine might trigger the original problem.

If you get the time I'd appreciate if you could run a test by putting a `Thread.sleep(1000);` here and see if that helps:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-visualizer/src/main/java/com/willwinder/ugs/nbm/visualizer/jogl/NewtVisualizationPanel.java#L153