winder/Universal-G-Code-Sender #2648: Auto Leveler

wmoscon1, 2024-12-02:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

I'm trying to use the Auto Leveler, but when it touches the machine table it gives an error such as probe failed, but the probe does the job of detecting when it touches the table. grateful.

[image]

### Operating System

Windowns 10

wmoscon1, 2024-12-02:
[image]

breiler, 2024-12-03:
I think that the probe issue has been fixed in the nightly build.

wmoscon1, 2024-12-03:
Would it be for version 10 (2.1.10)?

breiler, 2024-12-04:
It does, but I introduced a graphical bug where the visualizer hangs which should be fixed in the latest nightly build: https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

wmoscon1, 2024-12-16:
I was testing the version you gave me and it is also bad, as it gets stuck in the middle of making the piece and I end up losing the raw material. I believe you don't even have a stable version? Grateful

breiler, 2024-12-17:
What do you mean by "as it gets stuck in the middle of making the piece"? What happened? Did you get any errors?

wmoscon1, 2024-12-18:
Ask me a question, after I do the auto leveling, do I have to save the file and run the saved file with the auto leveling?

breiler, 2024-12-19:
The autoleveling should be applied to the loaded Gcode.

wmoscon1, 2024-12-20:
Ok, but for me to apply it, I have to export gcode, is that it?

wmoscon1, 2024-12-20:
In this case, I export gcode, but at the moment the auto leveler is running it finds the table and the PCB board, but when it goes to execute the gcode with the auto leveler applied, it either sinks too much or it doesn't even touch the table . grateful