winder/Universal-G-Code-Sender #2658: UGS sent previously opened file, despite showing current file

MikeBenza, 2024-12-11:
### Version

2.0.13

### Hardware / Firmware

GRBL 1.1

### What happened

I'm CNC carving a long piece of wood in 5 sections.  I've got 5 different NC files: section-1.nc through section-5.nc.  I'm working from 5 down to 1 just because that setup was easiest.  Sections 5 and 4 worked fine.  I remounted the piece for section 3 and just before I started it, realized I still had the section 4 file open.  I loaded section 3 and then ran it.

For some reason, it sent the section 4 file instead of the section 3 file.  The messages.log talks about loading section 3, but it sent section 4.

What it carved:
[image]

What the UI showed
[image]

[messages.log](https://github.com/user-attachments/files/18089515/messages.log)

[sections-3-4.zip](https://github.com/user-attachments/files/18089633/sections-3-4.zip)

Note: I'm using GRBL, but I'm not sure which version.

### How to reproduce

I haven't figured out how to reproduce this.

### Operating System

Mac M2 Sonoma 14.5

breiler, 2024-12-11:
Your version of UGS is ancient and I would recommend that you upgrade and try again:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

The other thing that I notice is a large amount of errors in the log. Do you perhaps have a offline controller connected to the machine?

MikeBenza, 2024-12-11:
I'll try upgrading when I get back to it in a few days.

I do have an offline controller.  I wasn't able to (quickly) figure out how to set up my z probe in UGS, and I preferred the continuous jog on the offline controller (I was much less likely to make a mistake in it than in UGS).

breiler, 2025-02-15:
Closing as we did not get further responses from user

closed 2025-02-15 by breiler