winder/Universal-G-Code-Sender #2660: Feching error

Tatarinyon, 2024-12-12:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

I placed ramps 1.4 on the arduino mega and connected it to the computer and installed the software  
I tried to open ugs and connect but it didn't work, I tried almost 10 different versions and it didn't work in any of them, always the same error;

*** Connecting to jserialcomm://COM6:9600
*** Fetching device status
>>> ?
*** Fetching device status (2 of 10)...
>>> ?
*** Fetching device status (3 of 10)...
>>> ?
*** Fetching device status (4 of 10)...
>>> ?
*** Fetching device status (5 of 10)...
>>> ?
*** Fetching device status (6 of 10)...
>>> ?
*** Fetching device status (7 of 10)...
>>> ?
*** Fetching device status (8 of 10)...
>>> ?
*** Fetching device status (9 of 10)...
>>> ?
*** Fetching device status (10 of 10)...
>>> ?
**** Connection closed ****

### Operating System

windows

breiler, 2024-12-12:
GRBL 1.1 uses baud 115200 by default, you have selected 9600.

breiler, 2025-02-15:
Closing as we did not get more information from user

closed 2025-02-15 by breiler