winder/Universal-G-Code-Sender #2661: Installer dmg will not run on my MacBook Air M2

stevenski, 2024-12-14:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

I downloaded the installer dmg from GitHub (twice) version "macosx-aarch64-ugs-platform-app-2.1.9"

First tried to install on Mac OS Sonoma 14.5 but dmg refused to run
Upgraded to Sonoma 14.7.2 but same thing - dmg simply does not start the install process.

Is this a known issue? Are there any fixes for someone who doesn't anything about using terminal or programming of any kind :)?

### Operating System

Mac OS Sonoma ver 14.5 and 14.7.2

breiler, 2024-12-14:
Did you follow the instructions on:
https://universalgcodesender.com/installing/

stevenski, 2024-12-14:
HI Joacim,

First of all thank you so much for your prompt answer!

Secondly, yes I did follow your video - as far as I could. Double clicking on the dmg file fails to launch the installer. Therefore I cannot proceed to the dragging and dropping etc

Best
Steven

breiler, 2024-12-14:
When you double click the DMG, you don't see the image as an mounted device in the Finder either?

stevenski, 2024-12-14:
Darn it! If ever there was a case study for keeping a tidy desktop, this is it ;)
Sorry - I didn't notice it there.

I may be a total noob to CNC but I am usually better than this! grrrr
Thank you and sorry for taking up your time like that. 

BTW would you have any recommendation for a starter, free CAD/CAM software for me to test the waters with? Sansmart 3018 prover v2 CNC together with the 5,5W laser optional module

Cheers
Steven

breiler, 2024-12-17:
I am not the right person to ask since I am biased towards Open Source tools.

If you are just getting started you might want to test the design features in UGS? They are intentionally very limited but could get you started:
https://www.youtube.com/watch?v=Q8dBg5UuUPQ

After that you might want to check out FreeCAD for both CAD/CAM or Kiri:Moto for CAM?

closed 2024-12-17 by breiler