winder/Universal-G-Code-Sender #2663: bad configuration file: GRBL

Guydevkh, 2024-12-16:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

I am using UGS for the last 6 months on a CNC Genmitsu 3018 Prover.  Using windows 10 on Lenovo ThinkPad,

 I started to use the Design feature in the last few days and everything seams to be OK.  I noticed that UGS can be connected to a laser engraver,  so I tried it on a Bachin 4050.  Seams to be working good.  Then today I worked with the laser engraver and the CNC, always disconnecting before switching,  got the message error:  **Bad configuration file: GRBL decimal processor use at least 4 decimal**

Closed the software, restart the computer and everything was OK, then problem came back,  download the latest version, restart the computer,  no improvements.

Now, I cannot use UGS with the CNC or the laser engraver.  Candle software is working good with CNC and laserGbl is working good with the laser engraver.

What could cause this problem and how to resolve it.

Thank you for your help

Regards
Guydevkh

breiler, 2024-12-17:
What does this setting say? It must be set to 4 or higher
[image]

breiler, 2025-02-15:
Closing as we did not get further responses from user

closed 2025-02-15 by breiler