winder/Universal-G-Code-Sender #2664: reset windows problem

Guydevkh, 2024-12-17:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

After reinstalling UGS this morning, everything was working good,  then I use the reset windows command then I lost all the windows, console, visualizer, etc.  Did try to add the windows back,  but it was not working properly, would not stay in the software screen
I remove the cache (following instructions from Bob's CNC) and reinstalled the software, still no windows as before,  got the following messages

WARNING: Cannot load even default layout, using internal predefined configuration

warning: cannot load window system persistent data, user directory content is broken. Reset to default layout...

Hopefully, you can help me again

Guydevkh

### Operating System

windows 10

Guydevkh, 2024-12-20:
I am including the messages file requested
[messages.log](https://github.com/user-attachments/files/18214870/messages.log)

Id di try to open each window from the menu, visualiser, console, etc. it does open but cannot change the size and position of the window,  if I try FLOAT,  I can change the size and position but does not stay on the main screen of UGS.

Guydevkh

breiler, 2024-12-21:
I think that your settings are corrupted, can you try deleting the folders:
`C:\Users\Guy\AppData\Roaming\ugsplatform` and `C:\Users\Guy\AppData\Local\ugsplatform`

breiler, 2026-10-31:
Closing as we did not get more information from user.

closed 2026-10-31 by breiler