winder/Universal-G-Code-Sender #2667: New Design not opening new design tab unless disconnected.

SW0rden, 2024-12-29:
### Version

2.1.9

### Hardware / Firmware

GrblHAL

### What happened

When attempting to use the UGS Designer, it doesn't open a new design tab when connected to the controller.  If UGS is in a disconnected state, then a new design tab will open.

### How to reproduce

Launch UGS and connect.  Choose New Design, and nothing happens, no new design tab is opened.  If you disconnect, or don't connect when first launched and then choose New Design, it works, and a new design tab is opened.
 

### Operating System

Windows 10 Pro

breiler, 2024-12-29:
Can you post your log file?

Reproduce this error, then go to the Tools menu and select "Open log directory". Attach the messages.log file to this GitHub issue.

SW0rden, 2024-12-29:
[messages.log](https://github.com/user-attachments/files/18268713/messages.log)

Thanks for your reply, here's the log file.

breiler, 2024-12-29:
Ah, this problem is already fixed and will be available in the next release.

You can give that a try already by downloading the nightly build from here:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

SW0rden, 2024-12-29:
Great, thanks!

closed 2024-12-29 by SW0rden