winder/Universal-G-Code-Sender #2673: ugs probe is not in the expected initial state

bvernham, 2025-01-09:
### Version

2.1.9

### Hardware / Firmware

GRBL 1.1

### What happened

I keep getting this error with UGS "probe is not in the expected initial state" which is very confusing becuase I have never had and issue with probing before.  I have probed with bCNC and with EASEL with zero issues.  With UGS I am getting (ALARM:4) Probe fail. Probe is not in the expected initial state before starting probe cycle when G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.  The machine is zeroed with everything hooked up (clip to the bit) and to the touch plate.  The machine probes down to the touch plate.  Makes contact, backs off, then set the alarm.  Probe cycle looks OK until the makes contact (like it should) then it fails?  Never had thie problem with any there g code sender.

### How to reproduce

See above.  Tried multiple times, same issue.

### Operating System

Windows 10

### Anything else

Every time I probe.  I wouldn't make this an issue if I had no probing experience with other applications.  But with Easel you can use machine inspector and the machine check out as expected.  The limit switches only set when you push them (as expected) and the probe is registered only when the touch plate is manually touching the bit.

bvernham, 2025-01-09:
>>> G10 L20 P1 Z0
>>> 
>>> G21 G91 G49
>>> G38.2 Z-90 F100

>>> 
ok
ok
ok
[PRB:0.000,0.000,-82.823:1]
>>> G91 G21 G0 Z2.0
>>> 
>>> G21 G91 G49
>>> G38.2 Z-90 F10
>>> 
ok
ok
ok
ok
ok
**** The communicator has been paused ****
An error was detected while sending 'G38.2 Z-90 F10': (ALARM:4) Probe fail. Probe is not in the expected initial state before starting probe cycle when G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered. Streaming has been paused.
ok
Error while processing response <ok>: An unexpected command was completed by the controller.

I tried a bigger #....

bvernham, 2025-01-09:
Now I am getting this:

>>> G10 L20 P1 Z0
>>> 
>>> G21 G91 G49
>>> G38.2 Z-90 F100
>>> 
ok
ok
ok
[PRB:0.000,0.000,-82.844:1]
ok
ok
>>> G91 G21 G0 Z3.0
>>> 
>>> G21 G91 G49
>>> G38.2 Z-90 F10
>>> 
ok
ok
ok
**** The communicator has been paused ****
An error was detected while sending 'G21 G91 G49': (ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked. Streaming has been paused.
[MSG:Reset to continue]

breiler, 2025-01-09:
Yes, you are probing down 90mm, which will go beneath you machines soft limits. You need to decrease the probe depth.

bvernham, 2025-01-09:
My limit was 90 but I guess that the probe is being done relative and not absolute so the machine is seeing 90 - 90 = -180.

I had the machine set at 110 which is real but now I have it set at -200 and the probe is working fine.  Does not see like a real safe way ot doing this to force set a large soft limit.

Can't the measure proble know where it is at and only need to probe down a small limit.

Also, why is the sender doing this:  G10 L20 P1 Z84.423

Should it just return to a safe Z?

Thanks

breiler, 2025-01-09:
I could probably improve the second probe cycle to just probe down the retract amount + a couple percent (instead of the full probe depth) to lower the risk of making the second probe exceed the soft limits.

The command `G10 L20 P1 Z84.423` sets the new work zero Z offset to the probed position (in machine space).

bvernham, 2025-01-09:
Shouldn't it be G10 L20 P1 Z1.61 (the thickness of the probe plate)?

breiler, 2025-01-09:
If it is not working as you are expecting, can you instead describe what is happening? Preferably with the console log and your current probe settings.

bvernham, 2025-01-09:
I am not sure why the spindle is going all the way back to the top of travel when the work to be done is at Z=0?  Should the spindle just go to a safe Z?

bvernham, 2025-01-09:
I put a macro in like this and it seems to do the trick:

G21 G91 G49;G38.2 Z-10 F50; G10 L20 P1 Z1.61;G4 P0.1;G0 Z5

breiler, 2025-01-10:
> I am not sure why the spindle is going all the way back to the top of travel when the work to be done is at Z=0? Should the spindle just go to a safe Z?

And what is safe Z? If the starting point was considered safe to start probing from, it makes sense to move back there as this will work for all probing scenarios (Z, XY, XYZ and Hole center). Otherwise we would need a setting for a "safe" X, Y and Z in all of these probing scenarios.

The algorithm for probing in UGS is like this:
* Probe towards Z with fast rate
* Probe touch - stop and retract
* Probe towards Z with slow rate
* Probe touch - record measured Z distance
* Retract Z to starting point
* Reset Z zero based on measured Z distance

https://github.com/user-attachments/assets/ac97a345-8847-4adb-88c3-83268601ef78

As I mentioned before, if you are using a large probe distance it might trigger a soft limit (if that is activated on the machine) alarm on the second slow probe. I will need to fix that, so that it probes the second time with just the retract distance.

> I put a macro in like this and it seems to do the trick

Great, than use that! It does things a little differently though:
* Probe towards Z
* Probe touch - stop
* Reset Z zero minus probe plate thickness
* Move to safe height Z5 (based on the new Z zero)

bvernham, 2025-01-10:
Ahh...so the trick is to start the probe from what is considered "safe" point, and then that is where it will return?

So home, jog to safe position, probe?

Let me give that a shot becuase that I think is the step I missed.

paulstraw, 2025-02-26:
I'm also running into this issue where an alarm is triggered after the initial retract. Increasing the retract amount as suggested does work, but then of course the probing takes much longer. Would it be possible to add a configurable delay after the initial retract as a workaround?

breiler, 2025-02-26:
I don't understand how a delay will help, either the probe is cleared or it is not.

Are you sure that you are retracting the probe enough to be disengaged? Does your machine or material flex while probing which could cause the probe to not become cleared?

paulstraw, 2025-02-26:
Yes, it's absolutely not making contact after the retraction, even at 1mm. I have to increase to ~5mm to reliably avoid this alarm though.

Maybe I'm misunderstanding the root issue here, but UGS doesn't report my probe as cleared for "a bit" after it's no longer making contact. Maybe 1-200ms? I'm testing this by manually touching the clip to the plate while looking at the DRO, so maybe this is just a UI thing? It seems to me that adding a delay after the retract would give the probe state change some time to be detected before beginning the next probing cycle, avoiding the ALARM: 4. My assumption was that's the same reason why a very large retract distance works.

breiler, 2025-02-26:
UGS polls the state of the controller by issuing status report commands. So this will cause some delay for the status to be presented in the UI. This is just presentation.

The controller handles these inputs with interrupts which should make it react very fast. But it will of course depend on if the inputs are protected with RC-filtering or other circuitry which is in place to handle false triggering due to noise.

What kind of hardware are you using?

You could try to verify your theory using a dwell command `G4 P<ms>`. 
I tried to whip up a gcode script example that you should be able to run on your machine. I have no way of verifying it so please be careful before running:
```
  G21             ; use metric
  G91             ; use incremental movement
  G38.2 Z-10 F100 ; probe down 10mm
  G0 Z1           ; move up 1mm
  G4 P50          ; dwell for 100ms
  G38.2 Z-2 F10   ; probe down 10mm
```

paulstraw, 2025-02-26:
Thanks! This does seem to confirm my theory. Adding a 100ms delay before the second probe reliably works, while commenting out the `G4` entirely triggers ALARM: 4.

This is a grbl machine (from MillRight). It's probably worth noting that `G4 P` often takes milliseconds in examples, but in grbl/LinuxCNC it's seconds. So, I used `G4 P0.1` for my 100ms delay.

paulstraw, 2025-02-27:
[image]

@breiler I've set up a branch with a working "Delay after retract" setting. Would you be interested in a PR?

breiler, 2025-02-27:
Nice, of course!

breiler, 2025-03-18:
There are now two fixes. One that should prevent the probing to exceed the soft limits. It will now retract and the probe the second time with the retract amount + 20%.

And a new setting from @paulstraw that will add a wait time between fast probe and slow probe.

closed 2025-03-18 by breiler