winder/Universal-G-Code-Sender #2686: Mac OS Sequoia V15.2. UGS doesn't open / malware notification

doubledutch2, 2025-01-24:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

Apologies as I'm not sure which version of UGS I use as I can't open it to check but:

1) I downloaded UGS on my mac yesterday from the main download section. I also tried the latest version on GitHub
2) I double-click the disk image and drag the application to my application folder
3) When opening the application, I get the following error and I don't know how to get around that.

[image]

### How to reproduce

See above

### Operating System

MacOS Sequoia 15.2

breiler, 2025-01-24:
This warning tells you that UGS did not pay for having the software signed through Apple and that by paying it is unlikely that it contains malware.

For those who don't trust our packaging it possible to build it yourself (even if it requires a bit of setup). See this file on which steps that are necessary:
https://github.com/winder/Universal-G-Code-Sender/blob/master/.github/workflows/release.yaml

If you do trust our packaging you should be able to run it using these instructions:
https://universalgcodesender.com/installing#macosx

closed 2025-01-24 by doubledutch2

breiler, 2025-01-24:
No worries! 

The user experience for this on MacOSX is bad... But we have tried as good as we can with the resources we have available.

doubledutch2, 2025-01-24:
@breiler I just discovered that with this MacOS what you suggest in the video actually doesn't work. It doesn't matter how often you try to open it, the error listed above comes up every time. There is a workaround that you may want to share with the larger audience which is below.

I feel better how, having potentially been able to help someone :-) 

[image]