winder/Universal-G-Code-Sender #2693: MPG mode switching and grblHAL

stevempotter, 2025-02-10:
### Description

I am new to UGS, so don't assume I know much here. I previously used IOsender for my MPCNC that is running on a Teensy 4.1 MCU. I would like to use UGS along with my home-made wireless pendant. The pendant is described in [this YouTube](https://youtu.be/lYjtsSwlnhk?si=Dk5VqP0TRIJkKn8k) vid. 
The CNC is running [grblHAL firmware](https://github.com/grblHAL), which has an MPG signal using an RX/TX UART line on the Teensy to ensure that only one thing is controlling the machine at a time, either the pendant or UGS on a laptop. 

Can UGS make use of this MPG flag functionality, described [here](https://github.com/grblHAL/core/wiki/MPG-and-DRO-interfaces)?

One of the benefits of grblHAL is it can go faster when a 32-bit processor like the Teensy is being used. 
Does UGS work natively for 32 bit-processing?

Maybe I just need to configure UGS differently to get my pendant to play nice with it?

breiler, 2025-02-15:
UGS will work with grblHAL for things that are compatible with the GRBL 1.1 protocol. Any extra stuff that grblHAL has implemented will probably not work. See https://github.com/grblHAL/core/wiki/Compatibility-level#compatibility-levels

UGS does not care if the controller is running on 8-bit or 32-bit, it is communicating with it through a serial connection. So this does not matter.

UGS does not read the MPG flag at this moment. If UGS receives the responses made by another device/connection it will confuse the internal work queue in UGS. I don't have a MPG capable grblHAL device available at the moment, so I can not test the behaviour myself.

stevempotter, 2025-03-02:
As I am eager to be able to use my pendant while UGS is running, I would be happy to test things for you to get an MPG flag reading functionality working in UGS. But I would need you (or anyone) to tell me what to test and how. The way I bench test things is to have a Teensy running grblHAL firmware but not connected to a CNC machine. (I have to spoof the endstops with jumpers). Maybe you could set up such a tester there?

The desired behavior is to use the MPG line to decide which of the two has control of the machine, UGS vs pendant.