winder/Universal-G-Code-Sender #2694: Toolbox window and 2nd row from top info boxes blank on windows 11 home edition laptop with 2.1.12

jjdike, 2025-02-11:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

The toolbox window and some of the other information windows are not populated.  This is on windows home edition 11 Dell laptop using latest UGS and openjdk version "1.8.0_442"

[image]

### How to reproduce

Start up UGS - with or without cnc attached

### Operating System

 This is on windows home edition 11 Dell laptop using latest UGS and openjdk version "1.8.0_442"

### Anything else

Happens every time.  I've tried several versions of Java.

OS Name	Microsoft Windows 11 Home
Version	10.0.26100 Build 26100
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	DESKTOP-OF12ICC
System Manufacturer	Dell Inc.
System Model	Inspiron 3501
System Type	x64-based PC
System SKU	0A25
Processor	11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz, 2419 Mhz, 4 Core(s), 8 Logical Processor(s)
BIOS Version/Date	Dell Inc. 1.34.0, 11/15/2024
SMBIOS Version	3.2
Embedded Controller Version	255.255
BIOS Mode	UEFI
BaseBoard Manufacturer	Dell Inc.
BaseBoard Product	0XGX0C
BaseBoard Version	A00
Platform Role	Mobile
Secure Boot State	On
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume1
Locale	United States
Hardware Abstraction Layer	Version = "10.0.26100.1"
User Name	DESKTOP-OF12ICC\user
Time Zone	Pacific Standard Time
Installed Physical Memory (RAM)	8.00 GB
Total Physical Memory	7.73 GB
Available Physical Memory	1.52 GB
Total Virtual Memory	19.7 GB
Available Virtual Memory	10.6 GB
Page File Space	12.0 GB
Page File	C:\pagefile.sys

breiler, 2025-02-11:
It looks like the settings has become corrupt. Try closing UGS then remove the folder: 
"C:\Users\[your username>]\AppData\Local\ugsplatform"

jjdike, 2025-02-11:
Thanks for the quick reply.  I tried this and the UI looks the same.

jjdike, 2025-02-11:
Yes that did it!  Thanks!

closed 2025-02-12 by breiler