winder/Universal-G-Code-Sender #2695: Sending file produces HOLD status

Michigan-dd, 2025-02-14:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

I use Carbide Create Pro to generate g-code then UGS to run. When I send the code it immediately returns a HOLD status. The output appears to continue but the HOLD .  I have run other g-code  this way with no issues. I changed the CC code generator from basic g-code to GRBL but the same HOLD problem continues. The machine is a Fox Alien Masuter 3S.  

### How to reproduce

Carbide Create Pro saves g-code.  Start UGS and open the CC file just made. Send to machine' HOLD status appears almost immediately. 

### Operating System

Windows 11

### Anything else

I am new to this machine but have experience with Laser Engraver  and have been a woodworker for many years.

breiler, 2025-02-14:
Can you attach the Gcode that you are trying to send

Michigan-dd, 2025-02-14:
I tried to add files to this but it tells me that .nc files are not allowed

breiler, 2025-02-14:
Send it to joacim.breiler@gmail.com

breiler, 2025-02-14:
The gcode contains the command `M0` which is used for pausing the stream. Probably to make the user check that the correct tool is used.
https://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m0-m1

```
G90
G21
(Toolpath:...)
M05
M0 ;T201
M03S20000
G0X-60.298Y-76.617
Z0.250
G1Z-0.508F254.0
X-60.265Y-76.816F1397.0
X-60.228Y-76.981
X-60.180Y-77.150
```

breiler, 2025-02-26:
I am closing this as it is working as intended

closed 2025-02-26 by breiler