winder/Universal-G-Code-Sender #2703: Crashes after opening

MrMattHumbert, 2025-02-27:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

Started the software after unzipping, as per the video. After the splash screen clears the main window opens for a moment then closes. I've cleared the .ugs folder from my home directory. 

### How to reproduce

Start the application process

### Operating System

Windows 11 Home

### Anything else

Thank you!

[messages.log](https://github.com/user-attachments/files/19014748/messages.log)

breiler, 2025-02-27:
The last messages in the log is about the graphics card and I assume that it is crashing after that.

It says that it has loaded OpenGL 4.6.0 and UGS uses the old OpenGL 2. This is usually not a problem as graphics cards are usually backwards compatible. You could try and upgrade your graphics drivers and see if that helps.

dmgualb, 2025-03-05:
I have exactly the same issue. I have some other information that may be usefull:
- I`m installing UGS in low end computers. The computer I had the problem is an Intel N4120 with 6Gb RAM - Win11.
- I`ve tried to erase cache (in AppData) and the issue continues.
- I've tried the Win64 binary, and also the "All platform" version, with the latest JDK from Oracle, and with OpenJDK 21 from Microsoft. The behavior is the same.
- The "Classic" version of UGS runs ok and doesn't crashes, but the "Visualizer" won't show.
- Reverting to older versions, the last UGSPlatform version that works ok is 2.1.9. From 2.1.10 on it crashes.
- If I run 2.1.9 and Hide the Visualizer Window, in sequence UGS 2.1.12 will start successfully, but will crash as soon as I try to show the Visualizer Window.
- In another low end computer, with ARM64 (Snapdragon 7c)  Win 11 and only 4Gb of RAM, 2.1.12 runs ok.

In summary: the isssue was introduced in version 2.1.10, and it is related to the Visualizer Window. It is independent of the the Java version used.

The error log:
[hs_err_pid75316.log](https://github.com/user-attachments/files/19079315/hs_err_pid75316.log)

Hellfighter323, 2025-03-13:
I just picked up a new Laptop to run my CNC and I'm having the same issue. Any updates???

breiler, 2025-03-14:
@Hellfighter323 what operating system are you using? If Windows, can you provide the output from DxDiag

Hellfighter323, 2025-03-27:
> [@Hellfighter323](https://github.com/Hellfighter323) what operating system are you using? If Windows, can you provide the output from DxDiag

breiler,

Not sure where the output is for DxDiag. Here's system info: 

[image]

dmgualb, 2025-03-27:
> 
> Not sure where the output is for DxDiag. Here's system info:
> 
To run DxDiag, you can type "DxDiag" at the Search Box of Windows. 

[image]

Then, use the "Save all information" button to create an output file.

Hellfighter323, 2025-03-27:
Let's see if this gets you what you need:

[DxDiag 03 27 25.txt](https://github.com/user-attachments/files/19490618/DxDiag.03.27.25.txt)

breiler, 2025-03-27:
Thanks @Hellfighter323, you are using a "Intel(R) UHD Graphics 6XX" as well.

Can you try the latest nightly build, I made a small adjustment the other day that disables some features in the visualizer if the graphics card is missing some features https://github.com/winder/Universal-G-Code-Sender/discussions/2730#discussioncomment-12612035.

The latest nightly build is available here:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

Hellfighter323, 2025-03-28:
Still doing the same thing. It opens for a few seconds then closes.

tapistefan, 2025-05-11:
I have same thing happening but I cant find logs. Only thing I can find is when i run it from console|

```
The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
WARNING: package com.sun.java.swing.plaf.gtk not in java.desktop
WARNING: package sun.awt.X11 not in java.desktop
Unrecognized option --clearcache
WARNING: A terminally deprecated method in java.lang.System has been called
WARNING: System::setSecurityManager has been called by org.netbeans.TopSecurityManager (file:/C:/Users/tapi9/OneDrive/Desktop/ugsplatform/platform/lib/boot.jar)
WARNING: Please consider reporting this to the maintainers of org.netbeans.TopSecurityManager
WARNING: System::setSecurityManager will be removed in a future release
```

I tried nightly, and i tried deleting cache, i tried reinstalling JDK 17. Its always same behavior while classic version runs.

breiler, 2025-05-12:
@tapistefan the original issue was for Windows 11. From the logs you provided it looks like you are running Linux. Please create a new issue.

tapistefan, 2025-05-12:
> [@tapistefan](https://github.com/tapistefan) the original issue was for Windows 11. From the logs you provided it looks like you are running Linux. Please create a new issue.

This is also windows 11, I am not on Linux.

breiler, 2025-05-12:
@tapistefan sorry, I didn't see the file path which makes it obvious that you are running Windows...

Is your machine also running a Intel UHD grapics card (see details above on how to retreive this).

Does it work with the latest nightly build?
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

tapistefan, 2025-05-12:
Thanks for responding so quickly! I tried latest nightly build and same thing is happening. I have NVIDIA RTX4090 GPU.

breiler, 2025-05-12:
Can you try and move the UGS folder to another folder? It is placed under OneDrive which could cause problems.

tapistefan, 2025-05-12:
I Dissabled OneDrive completely and all its backups. All paths are restored to normal (without OneDrive) and same thing is still happening. I also tried running it from another folder and even another drive and its still same thing
```
The launcher has determined that the parent process has a console and will reuse it for its own console output.
Closing the console will result in termination of the running program.
Use '--console suppress' to suppress console output.
Use '--console new' to create a separate console window.
Java HotSpot(TM) 64-Bit Server VM warning: Options -Xverify:none and -noverify were deprecated in JDK 13 and will likely be removed in a future release.
WARNING: package com.sun.java.swing.plaf.gtk not in java.desktop
WARNING: package sun.awt.X11 not in java.desktop
Unrecognized option --clearcache
WARNING: A terminally deprecated method in java.lang.System has been called
WARNING: System::setSecurityManager has been called by org.netbeans.TopSecurityManager (file:/D:/ugsplatform/platform/lib/boot.jar)
WARNING: Please consider reporting this to the maintainers of org.netbeans.TopSecurityManager
WARNING: System::setSecurityManager will be removed in a future release
```

RalfRalf1, 2025-05-26:
Hello all,

I have the very same problem with a rather old and small Notebook with N3350 Celeron. The application crashes before the UI opens starting with version 2.1.10. Version 2.1.9 is fine.
I tested the 2.1.14 version  on Linux Mint 22.1 Cinnamon edition and it works very well on the very same computer with N3350. 
So it seams to be only a problem of the Windows version. 

Thanks a lot for your effort to find and fix the issue.
Best Regards 
Ralf

breiler, 2025-05-26:
I think that we have reached the end of the line with the 3D graphics library that we are using. My best guess is newer graphics cards requires updated OpenGL versions. But on MacOSX we are stuck with an older version.

Some of you have had problems as of 2.1.10 where I started using vertex buffers. Technically this is a OpenGL 2.0 spec, but some graphics cards does not support it.  I could probably revert those changes but I doubt that this will fix it for all users.

If you are interested you can give the new JavaFX version a try. Going forward I will focus on developing this variant only. Look under the title "UGS FX" under the nightly version:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/nightly

HuubBuis, 2026-01-18:
I have tried another Atom tablet running W10. Seems I installed W10 as 32 bit version and UGS will not start.

Can I assist you in any way?
I have access to a  mini PC that has an I3-9100 processor and Intel UHD630 build in graphics card. My main PC has an AMD ryzen 5 processor.

I am located in the Netherlands

breiler, 2026-01-19:
I have ripped out the vertex buffer/shader rendering which I believe is causing the problem. It is available in the latest nightly:
https://github.com/winder/Universal-G-Code-Sender/releases/nightly

HuubBuis, 2026-01-19:
I tested the nightly build and it boots without a problem even without cleaning up the folders. Tried connecting and homing and that also worked. Will try CNC milling this afternoon.

I included the log
[UGS 2.0.Nb.messages.log](https://github.com/user-attachments/files/24711318/UGS.2.0.Nb.messages.log)

edit: I use a WiFi connection so maybe that is causing the "serial errors"