winder/Universal-G-Code-Sender #2705: No abre Universal Gcode Sender

chrslira, 2025-03-01:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

Buen día... Mi problema es que no abre el Universal Gcode Sender, se descarga correctamente y se extrae correctamente, pero al abrir la aplicación únicamente aparece la pantalla de carga, pero al momento de que parece que abrirá la aplicación se cierra completamente y no abre.

Anteriormente ya había trabajado con Universal Gcode Sender y no había presentado este problema, hasta que por motivos de espacio borre los archivos, los volví a descargar de la página y me apareció ese error, estoy buscando por todos lados pero no encuentro la solución.

Pd: Espero me puedan ayudar 

### Operating System

Windows 11

### Anything else

Mi programación CNC la llevaba a cabo mediante un Arduino uno, pero al momento de conectarlos, las coordenadas no respondían, en lugar de aparecer "libre" aparecía "alarm" y no me deja mover mi maquina

breiler, 2025-03-03:
If you have recently upgraded UGS and it is not working properly, you may need to remove cached files. The Cache is kept in a user directory called ugsplatform and deleting this directory will force UGS to recreate it. The location of the directory is as the following:
`C:/Users/[your username>]/AppData/Roaming/ugsplatform`

chrslira, 2025-03-04:
> Si ha actualizado recientemente UGS y no funciona correctamente, es posible que deba eliminar los archivos almacenados en caché. El caché se guarda en un directorio de usuario llamado ugsplatform y, al eliminar este directorio, UGS se verá obligado a volver a crearlo. La ubicación del directorio es la siguiente: `C:/Users/[your username>]/AppData/Roaming/ugsplatform`

Hola, ya lo intente y me sigue haciendo lo mismo, me aparece un error acerca de la versión de Java, pero se supone que la versión de UGS ya trae java incluído, aún así descargue distintas versiones de Java pero sigue sin abrirlo

[image]

chrslira, 2025-03-04:
> Si ha actualizado recientemente UGS y no funciona correctamente, es posible que deba eliminar los archivos almacenados en caché. El caché se guarda en un directorio de usuario llamado ugsplatform y, [...]

De igual manera justo después de borrar el cache, intenta abrir el software como se muestra en la siguiente imagen, pero lastimosamente se vuelve a cerrar y no deja mover nada solo se congela y se cierra 

[image]