winder/Universal-G-Code-Sender #2706: Machine Coordinates not showing up in DRO window V2.1.12

LorneRanger, 2025-03-03:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

Hello Everyone,

Just updated to V2.1.12 and not seeing any reference to the machine coordinates in the DRO window? Think it was working in V2.1.9 before the update. Using Windows 11 system V24H2.

Also a question regarding the Outline button for moving the machine around the model using the current depth. If I have a file loaded that has the x and y zero points in the center of the project it works fine and moves around the limits of it. It doesn't work if the x and y zero points are set to start at the lower left of the project. It will just move to the start point and stay there without doing anything.

Sorry if I did this wrong but I did have this posted in Discussions then thought maybe it should have been here in Issues? 

Thanks,
Lorne

### Operating System

Windows 11 - 22H2

breiler, 2025-03-03:
> Sorry if I did this wrong but I did have this posted in Discussions then thought maybe it should have been here in Issues?

Posting in the discussion was just fine, I have just been busy and haven't had the time to answer. =(

If you have recently upgraded UGS you may need to remove cached files. The Cache is kept in a user directory called ugsplatform and deleting this directory will force UGS to recreate it. The location of the directory is as the following:
`C:/Users/[your username>]/AppData/Roaming/ugsplatform`

Regarding the outline, I am not able to reproduce this problem. Do you have a gcode file you can share. I think that you need to .zip it before you can upload it to github.

LorneRanger, 2025-03-03:
Hello, 
Thanks for your reply. Not sure what I'm looking for as far as the cache file goes. Don't see anything cache related?

Attached a zip file with two Gcode files and a screen capture of my UGS Platform directory contents. One of the Gcode files starts from center the other from lower left. Outline works for the center one but not the Lower left.

Lorne :)

[Gcode Files.zip](https://github.com/user-attachments/files/19055588/Gcode.Files.zip)

breiler, 2025-03-03:
Shut down UGS, then remove or move the contents of the folder that you display here:
[image]

Thanks for the sample gcode, I managed to reproduce and fix the problem in  https://github.com/winder/Universal-G-Code-Sender/pull/2707. It is available in the latest nightly build.

LorneRanger, 2025-03-03:
Thanks again for your all your help! I have another machine I always use for testing new software updates. It runs under Windows 10 and it has the same problems with the machine coordinates not showing up in the DRO. I did as you suggested but made a BU copy of the ugsplatform directory first then deleted all the files in it. I had previously exported all the options out so I'd be able to import them afterwards as I've made some changes to the visualizer.

Downloaded and installed the latest nightly builds. It corrected the problem when using the outline function . . thanks. The problem of not seeing the machine coordinates in the DRO still exits though?

breiler, 2025-03-03:
Strange, this is the first time I hear of this specific problem. Can you go to the menu "Tools" / "Open log directory" and attach the file called "message".

LorneRanger, 2025-03-03:
Hi, I've attached two zipped log files and a screen capture for you. One is for the Win10 machine the other for the Win 11machine. Screen capture from the DRO Win11 machine.

[UGS Win10 Machine.zip](https://github.com/user-attachments/files/19061164/UGS.Win10.Machine.zip)
[UGS Win11 Machine.zip](https://github.com/user-attachments/files/19061165/UGS.Win11.Machine.zip)

[image]

LorneRanger, 2025-03-04:
Goodmornig! 
No worries . . so happy to see them back again, lol👍 I'd previously looked through all the menu's wondering if there was some kind of toggle. Guess it would have been easier just to have asked that?

Thanks again,
Lorne 😊

closed 2025-03-05 by breiler