winder/Universal-G-Code-Sender #2710: Error about arc segments when opening a file

token47, 2025-03-10:
### Version

2.1.9

### Hardware / Firmware

FluidNC

### What happened

When I open a file that contains G2/G3 commands, I immediattely get errors like:

[image]

The only way to unblock the GUI is to click OK on the dialog. My file has more than 2 thousand arc commands, so it is absolutely a pain to click OK 2k times to be able to use UGS.

The problem occurs as soon as I open the file, I don't need to "play" it. After I close all error messages (takes a few minutes of clicking), the file works normally. FluentNC processes the arcs normally. I don't think this is a problem of the interaction between UGS and the controller, but something on UGS itself when opening the file.

I have ARC Expansion plugin/extension disabled. Maybe this has something to do with the visualization instead?

Anyway, having to click OK 2 thousand times is not fun, even if the error is legitimate (which I don't think it is).

Any ideas what could be wrong?

### How to reproduce

Just open a gcode file with this:

```
G0 X57.60 Y226.78 Z3.00
G1 X59.08 Y232.59 Z0.00 F5000
G2 X65.14 Y230.97 R-3.200 F3000
G1 X63.51 Y225.20 Z3.00 F5000
```

My file is basically this repeated ~2k times with different coordinates.

### Operating System

Product Version: Universal Gcode Sender 20250108, Java: 17.0.8.1; OpenJDK 64-Bit Server VM 17.0.8.1+1, Runtime: OpenJDK Runtime Environment 17.0.8.1+1, System: Linux version 6.8.0-52-generic running on amd64; UTF-8; en_US (ugsplatform)

breiler, 2025-03-10:
Can you attach the whole Gcode file

breiler, 2025-03-10:
I would like to see the first couple of lines that sets up the initial state of the controller.

token47, 2025-03-10:
> I would like to see the first couple of lines that sets up the initial state of the controller.

This is the whole file. I just load this, send and the tool moves around. I generated this with some python code (the coordinate of each nail is known, I just go around them).

Before I actually run this, I use joystick in UGS to home, move around, zero machine on the right place, and then run this file.

token47, 2025-03-10:
I'm mostly relying on defaults here (I use absolute coordinates, milimeters, etc. All of that is default so I just don't set them. And, again, I don't actually use any tool (no need to activate spindle, etc.).

token47, 2025-03-10:
I just added a few initialiation gcode before the movements:

```
G21 ; millimeters
G90 ; absolute coordinate
G17 ; XY plane
G94 ; units per minute feed rate mode
; nail_index = 1, nail_number=197
G0 X103.49 Y209.85 Z3.00
G1 X106.14 Y215.23 Z0.00 F5000
G3 X100.47 Y217.94 R-3.200 F3000
G1 X97.96 Y212.49 Z3.00 F5000
```

The error still happens.

breiler, 2025-03-10:
I might found the problem, the visualizer does not like negative radius. I'll keep digging...

token47, 2025-03-10:
> I might found the problem, the visualizer does not like negative radius. I'll keep digging...

Indeed! If I replace all radius for positive numbers, it does not happen!

breiler, 2025-03-11:
I think that I managed to fix this in the latest nightly build and it looks like it is interpreted correctly:

[image]

The latest nightly build can be found here:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

token47, 2025-03-11:
Confirmed, it seems to work correctly now in todays nightly build. Before, not only I would get the error messages but also the arcs would not show in the preview, now I don't get any error and the arcs show correctly.

Again, thank you very much for fixing this.

closed 2025-03-11 by breiler