winder/Universal-G-Code-Sender #2721: Howdy!

LogrythmAtGmailDotCom, 2025-03-19:
Howdy!
Firstly let me thank everybody involved! 
My win laptop just died (15+ years old) and I was able to pick up where I left off with this not quite so old Mac laptop. 
Which is how I found out there is a Mac version lol (good old Java!).

All seems well, jobs completed etc, no issues. Except for these pesky exceptions:

(From the OSX Console)
"java.lang.ClassNotFoundException: com.willwinder.universalgcodesender.firmware.fluidnc.FluidNCController starting from ModuleCL@731a9cc4[com.willwinder.ugs.platform.gcode.editor] with possible defining loaders [ModuleCL@1601f65b[com.willwinder.ugs.platform.ugslib]] and declared parents [ModuleCL@470ca435[org.openide.loaders], ModuleCL@4814f1bc[org.netbeans.modules.editor.lib], ModuleCL@16a4f87c[org.netbeans.modules.csl.api], ModuleCL@2e26667[org.netbeans.modules.editor.lib2], org.netbeans.JarClassLoader@55a6c7a3, ModuleCL@1601f65b[com.willwinder.ugs.platform.ugslib], ModuleCL@d480266[org.openide.filesystems.nb], ModuleCL@26955674[org.openide.text], ModuleCL@650ff239[com.willwinder.ugs.platform.visualizer], ModuleCL@46d976a8[org.netbeans.modules.editor.mimelookup], ...11 more]
Caused: java.lang.NoClassDefFoundError: com/willwinder/universalgcodesender/firmware/fluidnc/FluidNCController
	at com.willwinder.ugs.nbp.editor.parser.errors.InvalidGrblCommandErrorParser.handleToken(InvalidGrblCommandErrorParser.java:66)
	at com.willwinder.ugs.nbp.editor.parser.GcodeParser.lambda$parse$0(GcodeParser.java:81)
	at java.base/java.util.ArrayList.forEach(Unknown Source)
	at com.willwinder.ugs.nbp.editor.parser.GcodeParser.parse(GcodeParser.java:81)
	at org.netbeans.modules.parsing.impl.TaskProcessor.callParse(TaskProcessor.java:598)
	at org.netbeans.modules.parsing.impl.SourceCache.getResult(SourceCache.java:230)
	at org.netbeans.modules.parsing.impl.TaskProcessor$RequestPerformer.run(TaskProcessor.java:775)
	at org.openide.util.lookup.Lookups.executeWith(Lookups.java:288)
	at org.netbeans.modules.parsing.impl.TaskProcessor$RequestPerformer.execute(TaskProcessor.java:702)
[catch] at org.netbeans.modules.parsing.impl.TaskProcessor$CompilationJob.run(TaskProcessor.java:663)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at org.openide.util.RequestProcessor$Task.run(RequestProcessor.java:1403)
	at org.netbeans.modules.openide.util.GlobalLookup.execute(GlobalLookup.java:45)
	at org.openide.util.lookup.Lookups.executeWith(Lookups.java:287)
	at org.openide.util.RequestProcessor$Processor.run(RequestProcessor.java:2018)"

Whatever it is, it keeps on tripping. So I'm reticent to run much more than test jobs atm.
Any tips appreciated..

[image]
[image]

_Originally posted by @LogrythmAtGmailDotCom in https://github.com/winder/Universal-G-Code-Sender/discussions/2713_

breiler, 2025-03-19:
These errors should now have been fixed in the upcoming release:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.1.13

LogrythmAtGmailDotCom, 2025-03-20:
Awesome. Thanks for letting me know :)

As I said earlier, I'm super happy UGS is truly cross-platform. 
I didn't imagine for a second that, after a laptop failure, I could continue as before. 
And on my preferred platform no less!

That never happens lol.

Cheers

closed 2025-03-20 by LogrythmAtGmailDotCom