winder/Universal-G-Code-Sender #2722: mvn install fails

andrewfernie, 2025-03-19:
### Version

2.1.12

### Hardware / Firmware

GrblHAL

### What happened

I would like to look at adding support for the grblHAL MPG flags so need to be able to build UGS. 

I cloned the repository from branch Master. Tried to execute mvn install but this fails.   Details below in the "How to reproduce" section.

### How to reproduce

sudo apt install maven  (Seemed to work OK)
git clone https://github.com/winder/Universal-G-Code-Sender.git
cd Universal-G-Code-Sender
mvn install

See  "apt install maven Log.txt" for details of maven installation.

See screenshot for mvn install errors, and "mvn install log.txt" for full details.

[apt install maven Log.txt](https://github.com/user-attachments/files/19352838/apt.install.maven.Log.txt)

[mvn install log.txt](https://github.com/user-attachments/files/19352839/mvn.install.log.txt)

[image]

### Operating System

Linux Mint 22 

### Anything else

Happens each time I try it. First try was March 18, but tried again today (March 19) in case there was a problem with the package being retrieved.

Operating system:
Distributor ID:	Linuxmint
Description:	Linux Mint 22
Release:	22
Codename:	wilma

Additional context
Maven: 3.8.7-2

Java:
openjdk version "21.0.6" 2025-01-21
OpenJDK Runtime Environment (build 21.0.6+7-Ubuntu-124.04.1)
OpenJDK 64-Bit Server VM (build 21.0.6+7-Ubuntu-124.04.1, mixed mode, sharing)

While my primary goal is to get this running on linux (logs above are for linux) I tried the same basic process on Windows 11 and ended up with the same error during the mvn install - the error messages were related to the same package.

While I started with a straight checkout of the Master branch, which looks like it is 2.1.13,  I also tried checking out tags  2.0.21, 2.1.12 and 2.1.0 with the same results. I only tried these other branches on linux, not Windows.

Also tried the same steps on Ubuntu 24.04 for the current 2.1.13 tag with the same result,

andrewfernie, 2025-03-19:
One other note: I forked the repository to do the tests described above. The day after the fork I received an email from github to say that the nightly build had failed. I hadn't intentionally asked for a build, but somehow it seemed to be setup automatically.,

[image]

breiler, 2025-03-20:
I got the same build error.

Pull the latest changes from master, it seems that jogamp has changed its domain from www.jogamp.org to jogamp.org which failed on a SSL certificate check.

The github actions will automatically run on any fork of the repository that will build and test the code and is a part of the project.

andrewfernie, 2025-03-20:
Thank you for your help. That fixed it and I can now install/build/run on my linux and windows machines.

closed 2025-03-20 by andrewfernie