winder/Universal-G-Code-Sender #2733: AutoLeveler window is empty

ves011, 2025-03-28:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

Autoleveler window is empty. No controls at all. Cannot set resolution, load height file, start scan... nothing - see the picteure attached
The log (attached) say "SEVERE [UGSEventDispatcher]: Could not dispatch the event SettingChangedEvent to the listener AutoLevelerTopComponent"

Feature used to work fine until today
I removed both appdata\local\ugsplatform and appdata\roaming\ugsplatform folders, restart UGS but no success. I did that several times

[image]

[messages.log](https://github.com/user-attachments/files/19507231/messages.log)

### How to reproduce

Starting today looks its a permanent issue.

### Operating System

Windows 11

### Anything else

Not sure abaut the version of UGS i use.
Its the latest version from github but cannot find in "about" any 2.x.xx

Product Version: Universal Gcode Sender 20250108
Java: 17.0.8.1; OpenJDK 64-Bit Server VM 17.0.8.1+1
Runtime: OpenJDK Runtime Environment 17.0.8.1+1
System: Windows 11 version 10.0 running on amd64; UTF-8; en_US (ugsplatform)

breiler, 2025-03-28:
Thanks for the report. Give me an hour and I'll see if can have a fix for this in place.

ves011, 2025-03-28:
@breiler Thanks!

breiler, 2025-03-28:
I added a check that will ensure that the zRetract percentage is not below the minimum value. It is available in the latest nightly build here: https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

If you succeed with the circuit board I'd love to see a picture!

ves011, 2025-03-28:
thx again
will give a try right now with the nightly build

ves011, 2025-03-28:
Same issue :(

Product Version: Universal Gcode Sender 20250328
Java: 17.0.8.1; OpenJDK 64-Bit Server VM 17.0.8.1+1
Runtime: OpenJDK Runtime Environment 17.0.8.1+1
System: Windows 11 version 10.0 running on amd64; UTF-8; en_US (ugsplatform)

However when i load the code sometime i get this error:

`java.lang.IllegalArgumentException: (minimum <= value <= maximum) is false
	at java.desktop/javax.swing.SpinnerNumberModel.<init>(Unknown Source)
	at java.desktop/javax.swing.SpinnerNumberModel.<init>(Unknown Source)
	at com.willwinder.universalgcodesender.uielements.components.PercentSpinner.<init>(PercentSpinner.java:37)
	at com.willwinder.universalgcodesender.uielements.components.PercentSpinner.<init>(PercentSpinner.java:49)
	at com.willwinder.ugs.platform.surfacescanner.ui.AutoLevelerPanel.initComponents(AutoLevelerPanel.java:88)
	at com.willwinder.ugs.platform.surfacescanner.ui.AutoLevelerPanel.<init>(AutoLevelerPanel.java:73)
	at com.willwinder.ugs.platform.surfacescanner.AutoLevelerTopComponent.initComponents(AutoLevelerTopComponent.java:151)
	at com.willwinder.ugs.platform.surfacescanner.AutoLevelerTopComponent.componentOpened(AutoLevelerTopComponent.java:108)
[catch] at org.openide.windows.WindowManager.componentOpenNotify(WindowManager.java:282)
	at org.netbeans.core.windows.WindowManagerImpl.notifyTopComponentOpened(WindowManagerImpl.java:1192)
	at org.netbeans.core.windows.Central.addModeOpenedTopComponent(Central.java:773)
	at org.netbeans.core.windows.ModeImpl.addOpenedTopComponent(ModeImpl.java:336)
	at org.netbeans.core.windows.LazyLoader.lazyLoad(LazyLoader.java:117)
	at org.netbeans.core.windows.PersistenceHandler.initModeFromConfig(PersistenceHandler.java:416)
	at org.netbeans.core.windows.PersistenceHandler.load(PersistenceHandler.java:179)
	at org.netbeans.core.windows.WindowSystemImpl.load(WindowSystemImpl.java:56)
	at org.netbeans.core.GuiRunLevel$InitWinSys.run(GuiRunLevel.java:200)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:136)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)`

breiler, 2025-03-28:
Try and shut down UGS again and delete this folder "C:\Users\[your username>]\AppData\Local\ugsplatform\Cache" and try open it again.

If it still doesn't work can I have a fresh copy of the message log file.

breiler, 2025-03-28:
Stupid bug (or stupid developer)...

I have another fix for you in the nightly build https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

ves011, 2025-03-28:
Is it a new build with the same name?

breiler, 2025-03-28:
Yes, it is a snapshot build that gets over written without any versioning (besides build timestamp)

ves011, 2025-03-28:
yep! this one works!!!
Don't close it until my pcb is ready :)
it might take 1hr or so

[image]

breiler, 2025-03-28:
Thanks for you patience sorting this out. It will certainly help others!

ves011, 2025-03-30:
Working, but still some improvements will be highly appreciated.
Will raise a feature request and attach the picture of the pcb there (because its related).
Thanks a lot for your outstanding support.

closed 2025-03-30 by ves011