winder/Universal-G-Code-Sender #2736: improve interpolation algorithm in autoleveler

ves011, 2025-03-30:
### Description

Autoleveler is a critical feature for us which are milling prototyping pcbs.
Doing these days a small pcb i faced a problem engraving lower right corner of my board. The bit didnt advance as wanted and the traces were not  isolated properly or not isolated at all. So i had to correct them by hand. You can see in the picture the faulty area.
It looks for me the way the interpolation algorithm did apply the height map for lower right corner is not ok.
Maybe you can have a look on this.
I attached the height map, original cnc isolation  file and autoleveled file after applying the height map to the original file.

[pump_controller-B_Cu.gbr_iso_combined_cnc - Copy.txt](https://github.com/user-attachments/files/19524569/pump_controller-B_Cu.gbr_iso_combined_cnc.-.Copy.txt)
[pump_controller-B_Cu.gbr_iso_combined_cnc-autoleveled - Copy.txt](https://github.com/user-attachments/files/19524568/pump_controller-B_Cu.gbr_iso_combined_cnc-autoleveled.-.Copy.txt)

[image]

breiler, 2025-03-30:
The algorithm for applying the height map is no different in the top right or lower left corners. It is either applied or it isn't.

Too me this sounds like the board is flexing during either probing or milling. Or the machine is loosing steps or have backlash. How is your board fixtured?

I am also curious what happened here, it looks like it lost steps in both X and Y direction?
[image]

breiler, 2025-03-30:
Going through the job - the lower left parts is not done in in go. Meaning that if it were a problem with backlash or missing steps it would be shown in the whole program. So I think that we can rule that problem out. 

https://github.com/user-attachments/assets/404c0d74-60ab-44a4-b6ed-e7f9bead569e

That leaves if the there is flex during probing/milling?

ves011, 2025-03-30:
@breiler Thanks again :)
The board is glued to the cnc bed and i did that with few hours before so i would say its a firm bond. As you can see the board is pretty small and 1.5mm thick. Unlikely to flex on ~3 cm with roughly 0.2mm (but not to exclude) between taking the height map and running the code.
About the area you asked, i corrected by hand the isolation. Thats why you see that ugly zone. I should take a picture before hand correction

breiler, 2025-03-30:
You don't happen to have saved the probe data (not just the autoleveled gcode)?

ves011, 2025-03-30:
yes i have it, thought is already attached

[pump_c.txt](https://github.com/user-attachments/files/19524721/pump_c.txt)

breiler, 2025-03-30:
[image]

The probed data for X50 Y10 and X60 Y10 is:
```
50.0 10.0 0.189
60.0 10.0 0.195
```

The desired depth in the original gcode is `Z-0.149` which when autoleveled in the selected rows in the screenshot would be `-0.149 + 0.189 => 0.04` which looks correct. To me it looks like the autoleveling is doing what it is supposed to.

So that suggests that the probe data is somehow wrong. I would double check how much play there is in the machine. Maybe run the probing at lower feed rates?

ves011, 2025-03-30:
Yes, you are right!
I checked myself in few other points and values seem ok.
Difficult to say how much play the machine has. will do the same height map several times to see the differences.
Nevertheless i appreciate taking you time in a sunday day :)

closed 2025-03-30 by ves011