winder/Universal-G-Code-Sender #2743: ugc sender direct close

ankitslov, 2025-04-02:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

i did use ugc sender it good but i am using now but direct close and not open have a some bug please help me.  you can please support and guide me how to fix this problems. thank you. ``

show this --> "An instance of the program seems to be already running with your user directory. Either a previous session of the program did not exit correctly, or another instance of the program is running on a different computer and using the same user directory." It further advises, "If another session of the program is running with the same user directory, please click Cancel to prevent the corruption of the user directory. If you are sure that no other instances of the program are running with your user directory, click OK to continue."

### Operating System

window 10

breiler, 2025-04-03:
Restart your computer and try to open UGS again. If it fails attach the log file from `C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log`

breiler, 2025-04-22:
Did not get response from user