winder/Universal-G-Code-Sender #2746: Exceptions thrown when Dealing with large GRBL.

dnikodem84, 2025-04-11:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

When loading "larger" Grbl ( 31 mb ) exceptions were raised. Size of Grbl may be unrelated, but an anti pattern was noticed. 

Specifically, control access from a "random thread". This will result in random visual glitches, potential deadlocks, and UI sluggishness. 

### How to reproduce

Load Large GRBL, watch traces in console. 

### Operating System

macOS Ventura

### Anything else

STACKTRACE: 

[INFO] WARNING [org.netbeans.core.windows.WindowManagerImpl]
[INFO] java.lang.IllegalStateException: Problem in some module which uses Window System: Window System API is required to be called from AWT thread only, see http://core.netbeans.org/proposals/threading/
[INFO] 	at org.netbeans.core.windows.WindowManagerImpl.warnIfNotInEDT(WindowManagerImpl.java:1812)
[INFO] 	at org.netbeans.core.windows.WindowManagerImpl.topComponentID(WindowManagerImpl.java:1454)
[INFO] 	at org.openide.windows.WindowManager.findTopComponentID(WindowManager.java:526)
[INFO] 	at org.netbeans.core.windows.model.TopComponentSubModel.getID(TopComponentSubModel.java:357)
[INFO] 	at org.netbeans.core.windows.model.TopComponentSubModel.containsTopComponent(TopComponentSubModel.java:253)
[INFO] 	at org.netbeans.core.windows.model.DefaultModeModel.containsTopComponent(DefaultModeModel.java:415)
[INFO] 	at org.netbeans.core.windows.model.DefaultModel.containsModeTopComponent(DefaultModel.java:886)
[INFO] 	at org.netbeans.core.windows.Central.containsModeTopComponent(Central.java:1538)
[INFO] 	at org.netbeans.core.windows.ModeImpl.containsTopComponent(ModeImpl.java:413)
[INFO] 	at org.netbeans.core.windows.WindowManagerImpl.findMode(WindowManagerImpl.java:625)
[INFO] 	at org.netbeans.core.windows.ModeImpl.dockIntoImpl(ModeImpl.java:240)
[INFO] 	at org.netbeans.core.windows.ModeImpl.dockInto(ModeImpl.java:153)
[INFO] 	at com.willwinder.ugs.nbp.designer.platform.PlatformUtils.lambda$openSettings$0(PlatformUtils.java:101)
[INFO] 	at java.base/java.util.Optional.orElseGet(Optional.java:364)
[INFO] 	at com.willwinder.ugs.nbp.designer.platform.PlatformUtils.openSettings(PlatformUtils.java:96)
[INFO] 	at com.willwinder.ugs.nbp.designer.platform.DesignerTopComponent.onSelectionEvent(DesignerTopComponent.java:188)
[INFO] 	at com.willwinder.ugs.nbp.designer.entities.selection.SelectionManager.lambda$fireSelectionEvent$3(SelectionManager.java:141)
[INFO] 	at java.base/java.util.concurrent.ConcurrentHashMap$KeySetView.forEach(ConcurrentHashMap.java:4706)
[INFO] 	at com.willwinder.ugs.nbp.designer.entities.selection.SelectionManager.fireSelectionEvent(SelectionManager.java:140)
[INFO] 	at com.willwinder.ugs.nbp.designer.entities.selection.SelectionManager.addSelection(SelectionManager.java:114)
[INFO] 	at com.willwinder.ugs.nbp.designer.gui.tree.EntitiesTree.valueChanged(EntitiesTree.java:92)
[INFO] 	at java.desktop/javax.swing.JTree.fireValueChanged(JTree.java:3019)
[INFO] 	at java.desktop/javax.swing.JTree$TreeSelectionRedirector.valueChanged(JTree.java:3520)
[INFO] 	at java.desktop/javax.swing.tree.DefaultTreeSelectionModel.fireValueChanged(DefaultTreeSelectionModel.java:650)
[INFO] 	at java.desktop/javax.swing.tree.DefaultTreeSelectionModel.notifyPathChange(DefaultTreeSelectionModel.java:1120)
[INFO] 	at java.desktop/javax.swing.tree.DefaultTreeSelectionModel.setSelectionPaths(DefaultTreeSelectionModel.java:306)
[INFO] 	at java.desktop/javax.swing.JTree.setSelectionPaths(JTree.java:1721)
[INFO] 	at com.willwinder.ugs.nbp.designer.gui.tree.EntitiesTree.onSelectionEvent(EntitiesTree.java:100)
[INFO] 	at com.willwinder.ugs.nbp.designer.entities.selection.SelectionManager.lambda$fireSelectionEvent$3(SelectionManager.java:141)
[INFO] 	at java.base/java.util.concurrent.ConcurrentHashMap$KeySetView.forEach(ConcurrentHashMap.java:4706)
[INFO] 	at com.willwinder.ugs.nbp.designer.entities.selection.SelectionManager.fireSelectionEvent(SelectionManager.java:140)
[INFO] 	at com.willwinder.ugs.nbp.designer.entities.selection.SelectionManager.addSelection(SelectionManager.java:114)
[INFO] 	at com.willwinder.ugs.nbp.designer.actions.TraceImageAction.lambda$actionPerformed$0(TraceImageAction.java:69)
[INFO] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
[INFO] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[INFO] 	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
[INFO] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
[INFO] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
[INFO] 	at java.base/java.lang.Thread.run(Thread.java:842)

breiler, 2025-04-11:
That log snippet is from the designer and not a gcode file.

Please include the whole log file.

dnikodem84, 2025-04-23:
The GRBL I am trying to output is rather nsfw so I am unsure if I should publish to github. 

Second Stacktrace from similar GRBL ( 31+ mb )

java.lang.IllegalStateException: Problem in some module which uses Window System: Window System API is required to be called from AWT thread only, see http://core.netbeans.org/proposals/threading/
	at org.netbeans.core.windows.WindowManagerImpl.warnIfNotInEDT(WindowManagerImpl.java:1812)
	at org.netbeans.core.windows.WindowManagerImpl.topComponentIsOpened(WindowManagerImpl.java:1440)
	at org.openide.windows.TopComponent.isOpened(TopComponent.java:517)
	at org.openide.windows.TopComponent.isOpened(TopComponent.java:508)
	at org.netbeans.core.windows.Central.addModeOpenedTopComponent(Central.java:755)
	at org.netbeans.core.windows.ModeImpl.addOpenedTopComponent(ModeImpl.java:336)
	at org.netbeans.core.windows.WindowManagerImpl.topComponentOpenAtTabPosition(WindowManagerImpl.java:1305)
	at org.netbeans.core.windows.WindowManagerImpl.topComponentOpen(WindowManagerImpl.java:1262)
	at org.openide.windows.TopComponent.open(TopComponent.java:473)
	at org.openide.windows.TopComponent.open(TopComponent.java:453)
	at org.netbeans.core.multiview.MultiViewCloneableTopComponent.ensureVisible(MultiViewCloneableTopComponent.java:371)
	at org.openide.text.CloneableEditorSupport.openAtImpl(CloneableEditorSupport.java:1934)
	at org.openide.text.CloneableEditorSupport.openAt(CloneableEditorSupport.java:1917)
	at org.openide.text.EditorSupportLineSet$SupportLine.show(EditorSupportLineSet.java:113)
	at org.openide.text.Line.show(Line.java:303)
	at com.willwinder.ugs.nbp.editor.FollowLineUpdater.updateCurrentLine(FollowLineUpdater.java:64)
	at com.willwinder.ugs.nbp.editor.SourceMultiviewElement.UGSEvent(SourceMultiviewElement.java:123)
	at com.willwinder.universalgcodesender.model.UGSEventDispatcher.lambda$sendUGSEvent$0(UGSEventDispatcher.java:66)
	at java.base/java.util.concurrent.CopyOnWriteArrayList.forEach(CopyOnWriteArrayList.java:807)
	at com.willwinder.universalgcodesender.model.UGSEventDispatcher.sendUGSEvent(UGSEventDispatcher.java:64)
	at com.willwinder.universalgcodesender.model.UGSEventDispatcher.commandComplete(UGSEventDispatcher.java:131)
	at com.willwinder.universalgcodesender.AbstractController.lambda$dispatchCommandComplete$5(AbstractController.java:841)
	at java.base/java.util.ArrayList.forEach(ArrayList.java:1511)
	at java.base/java.util.Collections$SynchronizedCollection.forEach(Collections.java:2131)
	at com.willwinder.universalgcodesender.AbstractController.dispatchCommandComplete(AbstractController.java:841)
	at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:787)
	at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:239)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:793)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.lambda$rawResponseListener$0(CommunicatorEventDispatcher.java:60)
	at java.base/java.lang.Iterable.forEach(Iterable.java:75)
	at com.willwinder.universalgcodesender.communicator.event.CommunicatorEventDispatcher.rawResponseListener(CommunicatorEventDispatcher.java:60)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.dispatchEvent(AsyncCommunicatorEventDispatcher.java:126)
	at com.willwinder.universalgcodesender.communicator.event.AsyncCommunicatorEventDispatcher.run(AsyncCommunicatorEventDispatcher.java:78)
	at java.base/java.lang.Thread.run(Thread.java:842)

dnikodem84, 2025-07-05:
A similar exception is thrown when attempting to import a SVG.

dnikodem84, 2025-07-05:
Thank-you for bringing my changes into trunk so quickly.

closed 2025-07-05 by dnikodem84