winder/Universal-G-Code-Sender #2752: Setting up soft limits fails on inverted axes

g3grau, 2025-04-14:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

I just went through the setup wizard in 2.1.13 and tried to set up soft limits on my Genmitsu 3030. I assume that the actual functions are ok, just the GUI has the +x and +y buttons disabled and I can only move Z out of the home position. I guess that the reason is that my machine has X and Y axes inverted and the button logic is screwed up (Z is not inverted and works as expected).

breiler, 2025-04-14:
Can you post the output from the `$$` and `$I` command in the terminal.

breiler, 2025-04-14:
And can you post a screenshot of the disabled jog buttons.

g3grau, 2025-04-14:
Everything looks ok except for the +-XY button enable/disable
```
*** Connected to GRBL 3.0
>>> $$
$0 = 10    (step pulse,usec)
$1 = 255    (step idle delay,msec)
$2 = 0    (stepport invert mask)
$3 = 3    (dirport invert mask)
$4 = 0    (stepenable invert,bool)
$5 = 1    (lims pin invert,bool)
$6 = 1    (probe pin invert,bool)
$10 = 3    (status report mask)
$11 = 0.010    (junction deviation)
$12 = 0.002    (arc tolerance,mm)
$13 = 0    (report inches,bool)
$20 = 1    (soft limits,bool)
$21 = 1    (hard limits,bool)
$22 = 1    (home cycle,bool)
$23 = 3    (homing dir invert mask)
$24 = 100.000    (homing feed,mm/min)
$25 = 500.000    (homing seek,mm/min)
$26 = 250    (homing debounce,msec)
$27 = 1.000    (homing pull-off,mm)
$30 = 12000    (maximum spindle speed,rpm)
$31 = 0    (minimum spindle speed,rpm)
$32 = 0    (laser mode enable,bool)
$100 = 800.000    (X axis pulse:step/mm)
$101 = 800.000    (Y axis pulse:step/mm)
$102 = 800.000    (Z axis pulse:step/mm)
$103 = 177.777    (A axis pulse:step/mm)
$110 = 2000.000    (X axis max rata:mm/min)
$111 = 2000.000    (Y axis max rata:mm/min)
$112 = 2000.000    (Z axis max rata:mm/min)
$113 = 1000.000    (A axis max rata:mm/min)
$120 = 20.000    (X axis acceleration:mm/s^2)
$121 = 20.000    (Y axis acceleration:mm/s^2)
$122 = 20.000    (Z axis acceleration:mm/s^2)
$123 = 20.000    (A axis acceleration:mm/s^2)
$130 = 300.000    (X aixs max travel:mm)
$131 = 300.000    (Y aixs max travel:mm)
$132 = 80.000    (Z aixs max travel:mm)
$133 = 9999.000    (A aixs max travel:mm)
ok
>>> $I
[VER:ARM32 V3.0.20230816:CUSTOM]
[OPT:VZL,35,254]
ok
```

[image]
[image]

breiler, 2025-04-14:
It is likely to do with the option "Homing force origin" in your controller (it is the Z in `[OPT:VZL,35,254]`). The homing direction should not matter.

This is a compile time parameter that makes the machine work in "positive machine space" instead of the default "negative machine space".  

Is this the default firmware of your machine or did you compile it yourself?