winder/Universal-G-Code-Sender #2758: ARM64 download archive appears to actually be for Mac OS

larry-athey, 2025-04-17:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

ARM64 archive appears to be the Mac OS version in a tar.gz archive. The JRE included in it is in a Mac OS type of "Contents" directory and the binary itself throws an error saying that it's for the wrong OS, likely a Unix executable.

Tested on an Orange Pi 3B running Armbian 6.1 (Ubuntu Noble)

breiler, 2025-04-17:
Which download link did you use?

It worked fine for me from the downloads page on https://universalgcodesender.com/download/

[image]

larry-athey, 2025-04-17:
I downloaded Linux ARM64 from https://github.com/winder/Universal-G-Code-Sender/releases/download/v2.1.13/linux-aarch64-ugs-platform-app-2.1.13.tar.gz

Here is what I'm talking about

[image]

breiler, 2025-04-17:
Aha, darn it! I'll try and fix it right away.

breiler, 2025-04-17:
Thanks for taking the time reporting and testing!

closed 2025-04-17 by breiler