winder/Universal-G-Code-Sender #2764: Not reset overrides at program start.

mikeoverbay, 2025-04-22:
### Description

Title says it all.

breiler, 2025-04-22:
Resetting overrides when starting a new stream were added in v2.1.12:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/v2.1.12

mikeoverbay, 2025-04-24:
I'd like to not reset overrides when it starts. Can we add a setting that defaults to set for those that don't want surprises?

breiler, 2025-04-24:
Lol, this was added from a request by another user that didn't want the surprise of keeping the overrides from the previous job.

mikeoverbay, 2025-07-01:
Yeah.. I figure that's why but its not normal behavior. lol
Add a check box to overides.. Reset on start. 
I cloned the repo. I've been studying it.
I am writing a module to interface with my pendant I build.
I am having issues finding some api sections.

I think I'll start a convo in the discussion area.
[image]