winder/Universal-G-Code-Sender #2765: Toolbox options greyed out.

CncKart0323, 2025-04-23:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

Having basically the same issue as described below from someone a couple years ago. But I don’t have any custom Marcos, it’s doing it with feed overrides and spindle overrides. 

I was on 2.1.12 but it started doing what I described so I saw there was 2.1.13 so I tried that, still doing it. I tried a nightly build also. 

Describe the bug
I have created some macros. If I add them to the "toolbox" they work fine, but then the next time I restart they are greyed out and unusable. If I go through settings, remove them and re-add them, they are normal again and work.

To reproduce
Steps to reproduce the behavior:

Machine -> Edit Macros
Click +Add
Provide a name and valid gcode
Click Try, confirm the operation
Click OK
Tools -> Options -> UGS -> toolbox
Find macro on the right panel, double click to add to the toolbox
Click OK
Confirm operation in the toolbox
Restart UGS
Do normal reconnection, homing, etc
Confirm CNC is responsive and normal
Macro appears in the toolbox but is unusable

breiler, 2025-04-23:
I can't reproduce the error. Can you take a screen shot of the whole application when this happens.

It would also help to get the log file from:
`C:/Users/[your username>]/AppData/Roaming/ugsplatform/var/log/messages.log`

CncKart0323, 2025-04-23:
Machine idle 

[image]

Machine running
[image]

[messages.log](https://github.com/user-attachments/files/19872093/messages.log)

closed 2025-04-24 by breiler

CncKart0323, 2025-04-24:
Ok thanks for your help. I don’t totally understand what I need to do to fix the issue for me.

breiler, 2025-04-25:
The fix is available in the latest nightly build:
https://github.com/winder/Universal-G-Code-Sender?tab=readme-ov-file#downloads

PR #2767 (2025-04-24): Fix override action serialization to be able to add it to the toolbox
Fixes #2765