winder/Universal-G-Code-Sender #2775: MM button not switching ti inches

reginaldStjohn, 2025-04-27:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

I am new using this but the "mm" button does not change its label to indicate wich units are active. I hit the button when it says "mm" and the units change in the coordinates box but the "mm" label stays on the button and the jog controlls, step size, feed rate, don't convert to cooresponding units.

### How to reproduce

Hit the "mm" button

breiler, 2025-04-28:
Can you take a screenshot.

reginaldStjohn, 2025-04-28:
Sure, When I get home tonight I will do that.  I will capture one before I click it an one after.

Randy

reginaldStjohn, 2025-04-29:
So I tried to reproduce it last night and was unable to do it. I had used it twice before and the button would not switch to say "inches" but last night it did everytime. 

If I see it a gain I will re-open the case

closed 2025-04-29 by reginaldStjohn