winder/Universal-G-Code-Sender #2788: Path Generation less than optional when creating designs in the designer

dnikodem84, 2025-05-14:
### Version

2.1.12

### Hardware / Firmware

GRBL 1.1

### What happened

When creating a Design in the designer and setting a cut depth which requires multiple slices. 

The resulting commands for the machine will "return to the safe height" between layers, This appears to be unnecessary.

Certain machines have an extremely slow z axis speed causing the resulting carve to take much longer than needed, especially when the carve depth is over 2 inches.

When slicing a shape we can assume that we have already traveled the previous layer prior to digging so we do not need to return to the tools "safe height" between layers.

I can see a use case where we need to 'clear debris' when drilling a hole of a similar size to the installed drill bit, so there is wisdom in raising the z axis prior to plunging. 

I propose an optimization in the designer where the resulting machine commands  for a path operation are modified to only lift the z axis to the height of the previous layer instead of the "safe height"

breiler, 2025-05-14:
Feel free to take a stab at it. 

I don't remember exactly how this was implemented. But I think it shares a segment type "SEAM" with the pocket operation. So look out for how the pocket operation behaves when there are holes in the geometry. This operation needs to clear the material before moving to another part of the pocket.

[image]

dnikodem84, 2025-05-15:
This will take some research on how to implement, Thank-you for the pointer to look into segment types. 

During Mill operations if there are no holes in the geometry then it would make sense to step down on the z-axis. 

I can imagine if milling a hole close to the size of your bit then you may want to actually raise the z-axis occasionally to clear debris / give the bit a chance to cool down / prevent fires.

dnikodem84, 2025-07-05:
I observed this operate correctly on my controller.  Please let me know if there is anything which I have overlooked.

closed 2025-07-07 by breiler